chirimen-oh/any-issues #5: Facebook グループ＆ Slack チーム改名

dynamis, 2016-04-15:
https://www.facebook.com/groups/mozopenhard/permalink/523344187853417/

https://www.facebook.com/groups/mozopenhard/ と http://mozoh.slack.com/ はコミュニティ名に合わせて変更が必要がと思います。例えば
* https://www.facebook.com/groups/chirimen/ + http://chrimen.slack.com/ 
* https://www.facebook.com/groups/chirimenopenhard/ + http://chirimenoh.slack.com/ 
* https://www.facebook.com/groups/chirimenorg/ + http://chrimenorg.slack.com/ 

何でも良いけどコミュニティ名に合わせて変えた方が良いかと思います。(説明分とかも)

gurezo, 2016-04-15:
超大雑把な計画書いてみました＠ぐんまー
コミュニティの皆様、適宜アドバイス＆修正をお願いします。

##  ロードマップ的なもの？
### 過度期
|月|週|作業内容|担当|備考|
|:-----:|------|------|------|------|
|4月|1W|経過|経過||
|4月|2W|経過|経過||
|4月|3W|経過|経過||
|4月|4W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|4月|5W|サイト切替作業|誰？|CHIRIMEN記念日イベント？<br />MDN<br />facebook<br />slack<br />github|
|5月|1W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|5月|2W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|5月|3W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|5月|4W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|5月|5W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|

### サイト切替作業期
|月|週|作業内容|担当|備考|
|:-----:|------|------|------|------|
|6月|1W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|6月|2W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|6月|3W|サイト切替作業|誰？|MDN<br />facebook<br />slack<br />github|
|6月|4W|サイト完全移行|誰？|CHIRIMEN移行宣言イベント？<br />MDN<br />facebook<br />slack<br />github<br />旧サイト閉鎖？|
|6月|5W|新サイト運用|誰？|MDN<br />facebook<br />slack<br />github<br />||

gurezo, 2016-04-15:
先程のロードマップは新しい issue を立てました。
内容がここでは無いと見直してみました。

twitter アカウントを作成しました。
[@chirimen_oh](https://twitter.com/chirimen_oh)

gurezo, 2016-04-16:
@dadaa 4/16 facebookグループ「MozOpenHard」=>「CHIRIMEN Open Hardware」変更 done.

gurezo, 2016-04-16:
これで、facebookグループのURL 変更できますかね？
http://alive-web.co.jp/tips/fb-3
http://socialmediaexperience.jp/4179

https://www.facebook.com/groups/mozopenhard/ 
=> https://www.facebook.com/groups/chirimenopenhard/

gurezo, 2016-04-16:
新しいslackチーム作成しました。
chirimenopenhardware.slack.com

dadaa, 2016-04-16:
facebook の URL ですけど，
https://github.com/chirimen-org/any-issues/issues/9
にも書いておきましたけれども，仕様上，一度しか変更できないみたいなのです．
これができないかどうか facebook に問い合わせしてみようと思います

dadaa, 2016-04-16:
が，夕方ぐらいからずっとフィードバックページが表示されないんです．とほほ．
https://www.facebook.com/help/feedback/

gurezo, 2016-04-16:
むむむむ・・・・。
新しく作って誘致居た方が良いですかね？
slack は、無理っぽかったので、作ってしまいました。
#17 参照

dynamis, 2016-04-18:
Slack は管理権限持っているユーザであれば改名できたのでは？
http://maplesystems.co.jp/blog/business/18298.html

gurezo, 2016-04-18:
これって、アカウント名だけでチーム名変更らしきものが
読み取れなかったので、新規作成しました。
※理解間違いの可能性は否めない（＞＜）

dynamis, 2016-04-18:
こちらが手順解説です:
https://get.slack.help/hc/en-us/articles/201663443-Changing-your-team-name-or-URL
Team Owners can rename their team or set a new web address at any time. Be sure to let all team members know about the change.

Owner が誰かは Admin 権限ある人が次のページを見れば分かるはず (多分赤塚さん)
https://mozopenhard.slack.com/admin

gurezo, 2016-04-18:
グンマーには、権限がないので無理っす。
どうしましょ？
念の為、こっちが作ったslackは削除しますか？
ここは、慌てないでお返事＆対応有無を聞いて、行動しまする。

dynamis, 2016-04-18:
needinfo: @dadaa

dadaa, 2016-04-18:
slack は，あたしやっておきますね

naobsd, 2016-04-29:
内容が #9 とかぶってるような気がします

gurezo, 2016-04-30:
#9 被っている＆新しいグループできたので、close

closed 2016-04-30 by gurezo