chirimen-oh/any-issues #10: RELEASEリポジトリを作る

MSakamaki, 2016-04-16:
### 置かれるもの

 + [ ] LICENCE
 + [ ] 回路図
 + [ ] B2G関連

satakagi, 2016-04-26:
ソフトには、release以外のリポがあって、ハードはreleaseのリポしかない理由はなぜ？

satakagi, 2016-04-26:
releaseには、ファームウェアのバイナリだけを置きますか？　ソースコードも置きますか？

satakagi, 2016-04-26:
ファーむぇあ書き込みツールとかは置きますか？

dadaa, 2016-04-26:
ハードはリリース以外のものってあります？
あるとすれば，Hardware リポジトリ作りましょうか

dadaa, 2016-04-26:
ソフトは，B2G を submodule として置けばよいかと思います
// 以前，議論して，そうしようとなったと思います
書き込みツールは置かなくてよいかと思います

satakagi, 2016-04-26:
ステッピングモーターとかの基板とか

dadaa, 2016-04-26:
あー，なるほど
ハードウェア単体のライセンスは CC で，CHIRIMEN ライセンスにはかからないですよね？

naokisekiguchi, 2016-04-26:
ビルドしたCHIRIMENイメージをどこかに置くべきだと思います。RELEASEレポジトリのreleasesにアップすべきかと思いましたがどうでしょう？

dadaa, 2016-04-26:
最終的にできる update.img が 300M を超えちゃっているので，
https://help.github.com/articles/conditions-for-large-files/
の制約にひっかかっちゃいそうです

naokisekiguchi, 2016-04-26:
@dadaa こちら https://github.com/chirimen-org/CHIRIMEN-tools/releases に過去のイメージがアップされています。zipファイルは150M前後ですが、ダウンロードしたファイルを解凍すると300Mを超えているので、おそらくzip圧縮後にアップすることは可能かと思われます。

dadaa, 2016-04-26:
@naokisekiguchi ほえー，ほいじゃ大丈夫そうですね!

satakagi, 2016-04-26:
自分でビルする人は まず少数派ですので、ビルドイメージは一番みんながアクセスしやすい場所に置いておくべきですね。
ちなみに、releaseはアクセスしやすい構造と導線作れます？

naokisekiguchi, 2016-04-27:
@satakagi 
https://github.com/chirimen-org/CHIRIMEN-tools/releases/tag/v2.5-20160109
こんな感じで特定のreleaseへのリンクが貼れます。

dadaa, 2016-04-29:
現状 release-candidate なので，明日，release に rename しましょう

closed 2016-04-30 by dadaa