chirimen-oh/any-issues #17: タスクリスト

gurezo, 2016-04-16:
超大雑把な計画的なモノを書いてみました＠ぐんまー
コミュニティの皆様、適宜アドバイス＆修正をお願いします。

### タスクりすと #5 より引用
 - [ ] CHIRIMEN | MDN
      ※ 扱いはどうでしたっけ？
 - [x] facebook group 変更 mozOpenHard => CHIRIMEN Open Hardware  
      ※グループ名は変更済み　4/16 @dadaa 
      ※urlは未変更
 - [x] slack 作成 CHIRIMEN Open Hardware  
      ※ https://chirimenopenhardware.slack.com 作成 @gurezo done. 
 - [x] twitter アカウント作成 @gurezo 04/16
 - [ ] CHRIMENアイコン作成  
      ※現状は、foxfoodイメージから切り抜き活用中
 - [ ] [tumblr：mozOpenHard](http://is.gd/5hzvBf)  
      - [ ] 上記サイト削除タイミング 
      - [ ] 移行サイト[https://chirimen-org.github.io/](https://chirimen-org.github.io/) 作成中 @MSakamaki  
 - [ ] [Fabble - ChirimenEdu](http://fabble.cc/chirimenedu)　　
      - [X] 現行のまま
      - [ ] B2G ver 2.5 or ver 2.6 で、再動作検証が必要 @naokisekiguchi

naobsd, 2016-04-29:
> 一般用Google Group作成 @gurezo 04/16 done.
>  スタッフ用Google Group作成 @gurezo 04/16 done.

https://groups.google.com/forum/#!forum/mozopenhard はどうなりますでしょか

gurezo, 2016-04-29:
これは、今さっき存在を知ったので、現時点でのオーナー権限のある方にお願いします。
ただ、そこでの議論は、残した方が良いとは思います。

dadaa, 2016-04-30:
CHIRIMEN | MDN の問題を除いて，issue 登録があるので，CHIRIMEN | MDN を issue とし，これを閉じます

closed 2016-04-30 by dadaa