chirimen-oh/any-issues #8: examplesリポジトリを作る

MSakamaki, 2016-04-16:


naokisekiguchi, 2016-04-18:
これ、どんな感じでしょう？
@MSakamakiさんのpolyfill ( https://github.com/club-wot/WebGPIO )を使って手元で動作確認とれたので、何かベースがあればfabbleのサンプルを合わせて作っていきますが。

MSakamaki, 2016-04-18:
@naokisekiguchi 
リポジトリ作りました。
https://github.com/chirimen-org/examples

あと、動作は実機で確認してないですがLED点滅のサンプルを仮であげてみました。
（問題有るなら消してOKです。）

naokisekiguchi, 2016-04-19:
@MSakamaki さきほど微修正したpull request出しました。

それと、これは好み分かれそうなんですけど、bower前提にしますか？使ったら便利なんですけどCHIRIMEN始めようってときにnpmとbowerのインストールが必要になるのはどうだろうと思いまして。
jsにライセンス情報を追記して直接含めるのはどうでしょう？他の人の意見も欲しいですね。

MSakamaki, 2016-04-19:
@naokisekiguchi 
> jsにライセンス情報を追記して直接含める

現時点では梱包しても問題ないですが、後々の事を考えるとpolyfillは別に出来た方が良いかなと。
polyfillを含めてしまう場合のデメリットとして、バグや仕変に追従出来なくなってしまう事が懸念されるので、最新のpolyfillに追従する方法のほうが個人的に良いかなと考えてました。

他の手段としては

 1. CDNに登録する
  + CHIRIMEN本体が常にInternetに接続している必要がある。
 1. 固定URLからzipでダウンロード
  + 導入手順が多少煩雑になる。
 1. `git submodule`化する
  + ぱっと思いつく限りコレが最善か・・・？

他に手段がないかちょっと考えたいですね。

naokisekiguchi, 2016-04-27:
@MSakamaki まだpolyfillも安定していないということで、別に分けてサンプルを作っていきましょうか。他にいい手段が出てくるまでbowerでインストールする形で作ります。

もしくはcloneした時点で使えるようになっているbranchが切られてもいいのかもと思いましたが。

MSakamaki, 2016-04-28:
@naokisekiguchi 
exampleまーじしときました、動作確認ができないの 自動テストとかでの品質チェックできないのが地味につらいですね。

closed 2016-04-30 by dadaa