chirimen-oh/any-issues #9: mozほげほげのリネーム、もしくは消す

MSakamaki, 2016-04-16:
+ [ ] facebook
 + [ ] slack
 + [ ] tumblr
 + [ ] twitter

dadaa, 2016-04-16:
facebook のグループ URL の変更は１回だけしかできない仕様のようで現在はもとのまま．facebook に変更できないか問い合わせてみます．

naokisekiguchi, 2016-04-18:
twitterは新アカウントの作成（https://twitter.com/chirimen_oh）と旧アカウント（https://twitter.com/MozOpenHard）への移行促進メッセージの投下を完了しました。あとはしかるべきタイミングで旧アカウントを消せば完了かと。

dadaa, 2016-04-18:
slack のリネーム完了．
chirimen-oh.slack.com
(アンダースコアは使えませんでしたのー)

dadaa, 2016-04-19:
Facebook さんからお返事きましたけど，やっぱり URL の変更は現在１回しかできず，個別対応もごめんなさい．だそうです．どしましょね

naokisekiguchi, 2016-04-19:
@dadaa すでに一度変更していてもう変更できないということですか？

MSakamaki, 2016-04-19:
@dadaa 
グループ削除はできるんですっけ？
新グループを作って引っ越し、２〜３ヶ月後に旧グループ削除が最善じゃないでしょうか？

dynamis, 2016-04-19:
@naokisekiguchi URL は一度変更しているというか、数字だけだったものを文字列に設定したら以後変更ができないというものだそうです。これは完全に想定してなかったですね。むぅ。

https://www.facebook.com/help/206871819351594
> 注: ウェブアドレスおよびメールアドレスは、カスタマイズすると変更できません。

dadaa, 2016-04-19:
@MSakamaki 現在 400 人ほどのメンバーがいるので，それをどう思うかが悩みどころです．．削除自体はメンバーが 0 になったらできるようです．

gurezo, 2016-04-20:
@dadaa @MSakamaki 
## 下記案如何でしょう
- グループ名と意図的にmozOpenHardへもどす
- CHIRIME Open Hardwareの新しいグループメッセージを作る
-  固定投稿に新ページへの参加とmozOpenHardページの退会を呼び掛ける
-  固定投稿にmozOpenHardページの閉鎖期日（日時）を設ける
- 期日を過ぎても残っている方は個別連絡＆依頼／各個撃破する。

これ位しか、思い当たりません・・・ｍ（＿＿）ｍ
すみません。

dadaa, 2016-04-23:
@gurezo さん，遅くなってしまってごめんなさい．

こちら，みなさま，どう思われますか？？
Facebook グループのリネーム案と新規案で，それぞれメリット/デメリットを挙げてみます．
リネーム案：
・メリット：現在のコミュニティメンバーを引き継げる
・メリット：同様にコンテンツも引き継げる
・デメリット： URL が moz る．
新規案：
・メリット：URL が綺麗になる
・メリット：コンテンツの整理ができる
・デメリット：過去の議論などを取り込めない

gurezo, 2016-04-23:
@dadaa さん、お返事有難うございます。

個人的には、大人の事情と書いてライセンスと読むのが原因で
揉めなければ、どちらでも良いと思います。

「・デメリット： URL が moz る．」が**_過去の歴史の経緯／遺産ですよ！_**

上記のようにぶった切って／言いきって、**_ライセンス問題に発展しなければ、_**
リネーム案が一番平和／手間が少ないとは思います。

揉める要素は無いに越した事はないのですが・・・。

MSakamaki, 2016-04-25:
@dadaa 
@gurezo さんのと同じく、URLについてmozillaから問題無いって回答が貰えればリネームで、カエテ！って言われたら新規案で良いと思います。

dadaa, 2016-04-26:
Facebook の URL，やっぱり問題になりそうなので，新規グループになりそうな予感です．．

naobsd, 2016-04-29:
#17 に書いちゃいましたけどこっちが正しそうなので

https://groups.google.com/forum/#!forum/mozopenhard どうしましょ

gurezo, 2016-04-29:
これは、今さっきmozOpenHardのGoogle Group の存在を知ったので、
現時点でのオーナー権限のある方にお願いします。
ただ、そこでの議論は、残した方／引っ越しが良いとは思います。

gurezo, 2016-04-30:
新しいCHIRIMEN Open Hardware の URLが変更されたら
facebook にチェック付けましょうか。

グループ移動の案内はポストした方が良いですか？

gurezo, 2016-04-30:
#35 に文面記載しました。
レビューお願いします。

gurezo, 2016-05-01:
mozOpenHardのGoogle Groupが削除されていたようなので、追記

dynamis, 2016-05-13:
Google Group は削除ではなく移動されてるはず: 念のため。
https://groups.google.com/forum/#!forum/chirimen_oh
既存メールアドレスに投稿しても届かないと思うので、変更したことの案内 (ついでに色々告知) は流しておいた方が良いかも。

naokisekiguchi, 2016-05-13:
今、
https://groups.google.com/forum/#!forum/chirimen_oh
と
https://groups.google.com/forum/#!forum/chirimen-oh
がありますが、どちらを正規として使っていく感じですか？

dynamis, 2016-05-13:
_oh! -oh!

投稿数と登録者数的には -oh の方が良さそうに見えます。
https://groups.google.com/forum/#!forum/chirimen-oh 
その場合、名称は _oh に揃えるなら今の _oh を改名の上で -oh を _oh にという流れになりますね。

naokisekiguchi, 2016-05-13:
なにげにgithubもslackも-ohなんですよね。
_ohはtwitterだけ？

_ohのメンバーを-ohに誘導して現状の_ohはcloseという形はどうでしょう？
（だいたいメンバーは被っていると思いますが）

gurezo, 2016-05-13:
twitterは、-使えないので、やむなく_にしました。

gurezo, 2016-07-02:
https://groups.google.com/forum/#!forum/chirimen_oh
#49 削除 done.

naobsd, 2016-07-08:
#97 https://travis-ci.org/MozOpenHard こんなものが

MSakamaki, 2016-07-08:
> #97 https://travis-ci.org/MozOpenHard こんなものが

MozOpenHard/gecko-dev のCIで使われてたのを見た事有ります。
gecko-devの引っ越しとともに中身は空っぽになってるので多分大丈夫かと。

naobsd, 2016-07-08:
@MSakamaki 大丈夫、というのは？

必要ないなら消す、必要なら名前を変える（消して作りなおす）、というアクションが必要、かと思っているのですが、例えばgithubのMozOpenHardに連動していて一緒に消えるから大丈夫、とかいう事なんでしょうか？(そもそもこれがどういうサービスなのか理解していません...)

MSakamaki, 2016-07-12:
@naobsd 
TravisはCI系サービス(BambooとかJenkinsのsaas版)です。
プルリク後の自動テストとか、masterブランチやタグ切ったら自動リリースとかするために使う奴です。
(chirimenのHPでも使ってます。)

基本名前の変更はできなく、githubをfookする形で生成されるURLです。
なので、MozOpenHeard -> chirimen-oh と言う名前の変更してれば消えていたかもしれないですが、今回２つ作って中身を移動したので残っていると言う状況です。

大丈夫というのは、唯一使ってたのはgecko-devリポジトリのみっぽかったので実質MozOpenHardが空っぽなので、githubのMozOpenHardを消せばそのうち消えると思います。
どのタイミングでURLがクリアされるのは不明ですが、以前自分が戯れに作ったTravisのURLは気がついたら消えてました（１年くらい？）
すぐにURLを消したい場合はTravis社に問い合わせないといけないかも。

dadaa, 2016-07-14:
遅くなりましたが tumblr は念のため削除はせずに、パスワード設定しておきました。

naobsd, 2016-08-08:
まだ消さなければいけないものが残っている認識です。

- [ ] https://github.com/mozopenhard
- [ ] https://developer.mozilla.org/en-US/docs/Mozilla/B2G_OS/Board_guide/CHIRIMEN
- [ ] https://developer.mozilla.org/ja/docs/Mozilla/B2G_OS/Board_guide/CHIRIMEN

https://travis-ci.org/MozOpenHard は github と連動して消えるはず?

naobsd, 2016-08-09:
https://github.com/chirimen-oh/any-issues/issues/110#issuecomment-238410278 に関して、個人的にはMozOpenHard時代の議事録・資料まで無理に直す必要はないと思います。(記録は記録としてそのままでよいかと)

現在も参照する必要のある情報については、chirimen.org等に掲載する際に適宜文面を修正すればよいと思います。

dynamis, 2016-08-10:
MDN ページの移動は #117 で別途やるとして TODO チェック埋めました。

gurezo, 2016-08-10:
08/10のミーティングで、Google Groupのメーリングリストを削除する事が決定。
#120