chirimen-oh/any-issues #19: slack 参加フォームを作る

gurezo, 2016-04-23:
## slack の自由参加フォームを作る
- 現在は招待しないと参加出来ない
- グーグルのフォームを使ってやる らしい
- 要調査
- 4/30 以降のイベントではここから自由参加が出来る様にする

gurezo, 2016-04-23:
### 招待方法
#### 下記のいずれかで出来る？
- [公開Slack用 自動招待フォーム by GAS](http://qiita.com/atusi/items/c2669add8038c43efff3)
- [Arukas.io で Slack 自動招待フォーム](http://qiita.com/pottava/items/cb3dad247d8e5778008e)

MSakamaki, 2016-04-25:
slackを使っている理由を良く知らないのでアレですが。
自由参加チャットとしてしか使っていないのならgithubアカウントさえ有れば参加できる https://gitter.im (github用チャット)もあります。
欠点としてはgithubアカウントを作らないと行けないくらいで、それ以外はsluckより判りやすく、エンジニア向けのチャットツールだと思ってます。
(githubマークダウン使えるし、githubのissuesと連携できるので)

gurezo, 2016-04-25:
そうゆう意味では、**slackの利用スタンスって自分も理解していませんでした・・・。**

今回の投稿は、招待する度にちょっと面倒だから自由に参加できるようにしたら？！
そんな具合で、@dadaa さんとお話になりました。

また、ng-japan の slackは、dooe keeper から自由参加出来たので
同じように出来るのかな？という背景も有りました。

----

### 整理したい事
- slack の利用スタンス
- 自由参加にするかどうか  
  ※上記が決まれば、自ずと決まる？
- ng-japanは、[https://ng-japan-invite.herokuapp.com/](https://ng-japan-invite.herokuapp.com/) で自由参加を募っていた。  
  - コミュニティのどなたかが作成対応した？  
  - door keeper の一斉送信メールで配信した。

gurezo, 2016-07-19:
SlackinでSlackに誰でも参加できるチームをつくる
http://qiita.com/r7kamura/items/0b0a2a2d536907816b0d

上記で作成完了
招待リンク
https://chirimen-oh.herokuapp.com/

誰か、新規でslackに加入確認が出来たらcloseする

gurezo, 2016-07-20:
現在参照不能になりました。
アプリ設定を失敗し、アカウント削除を誤ってしてしまった為。

gurezo, 2016-07-20:
アカウントを再作成し、デプロイしましたが
まだApplication Error発生しています。

gurezo, 2016-07-20:
```
-----> Node.js app detected
-----> Creating runtime environment
       
       NPM_CONFIG_LOGLEVEL=error      <== ここでエラー発生している？
       NPM_CONFIG_PRODUCTION=true
       NODE_ENV=production
       NODE_MODULES_CACHE=true
-----> Installing binaries
       engines.node (package.json):  >=0.12.0
       engines.npm (package.json):   unspecified (use default)
       
       Resolving node version >=0.12.0 via semver.io...
       Downloading and installing node 5.11.1...
       Using default npm version: 3.8.6
-----> Restoring cache
       Skipping cache restore (new runtime signature)
-----> Building dependencies
       Installing node modules (package.json)
       slackin@0.11.3 /tmp/build_4270bdebd46b993115da37a09caee77a/rauchg-slackin-75c7822
       ├─┬ args@1.3.0
       │ ├── camelcase@3.0.0
       │ ├─┬ chalk@1.1.3
       │ │ ├── ansi-styles@2.2.1
       │ │ ├── escape-string-regexp@1.0.5
       │ │ ├─┬ has-ansi@2.0.0
       │ │ │ └── ansi-regex@2.0.0
       │ │ ├── strip-ansi@3.0.1
       │ │ └── supports-color@2.0.0
       │ ├─┬ loud-rejection@1.6.0
       │ │ ├─┬ currently-unhandled@0.4.1
       │ │ │ └── array-find-index@1.0.1
       │ │ └── signal-exit@3.0.0
       │ ├── minimist@1.2.0
       │ ├── pkginfo@0.4.0
       │ └── replaceall@0.1.6
       ├─┬ babel-polyfill@6.3.14
       │ ├── babel-regenerator-runtime@6.5.0
       │ ├── babel-runtime@5.8.38
       │ └── core-js@1.2.7
       ├─┬ body-parser@1.10.2
       │ ├── bytes@1.0.0
       │ ├── depd@1.0.1
       │ ├── iconv-lite@0.4.6
       │ ├── media-typer@0.3.0
       │ ├─┬ on-finished@2.2.1
       │ │ └── ee-first@1.1.0
       │ ├── qs@2.3.3
       │ ├── raw-body@1.3.2
       │ └─┬ type-is@1.5.7
       │   └─┬ mime-types@2.0.14
       │     └── mime-db@1.12.0
       ├─┬ debug@2.1.1
       │ └── ms@0.6.2
       ├── email-regex@1.0.0
       ├─┬ express@4.11.0
       │ ├─┬ accepts@1.2.13
       │ │ ├─┬ mime-types@2.1.11
       │ │ │ └── mime-db@1.23.0
       │ │ └── negotiator@0.5.3
       │ ├── content-disposition@0.5.0
       │ ├── cookie@0.1.2
       │ ├── cookie-signature@1.0.5
       │ ├── escape-html@1.0.1
       │ ├─┬ etag@1.5.1
       │ │ └── crc@3.2.1
       │ ├── finalhandler@0.3.3
       │ ├── fresh@0.2.4
       │ ├── merge-descriptors@0.0.2
       │ ├── methods@1.1.2
       │ ├── parseurl@1.3.1
       │ ├── path-to-regexp@0.1.3
       │ ├─┬ proxy-addr@1.0.10
       │ │ ├── forwarded@0.1.0
       │ │ └── ipaddr.js@1.0.5
       │ ├── range-parser@1.0.3
       │ ├─┬ send@0.11.0
       │ │ ├── destroy@1.0.3
       │ │ └── ms@0.7.0
       │ ├─┬ serve-static@1.8.1
       │ │ └─┬ send@0.11.1
       │ │   └── ms@0.7.0
       │ ├── utils-merge@1.0.0
       │ └── vary@1.0.1
       ├── hostenv@1.0.1
       ├── opentype.js@0.4.4
       ├─┬ socket.io@1.4.8
       │ ├─┬ debug@2.2.0
       │ │ └── ms@0.7.1
       │ ├─┬ engine.io@1.6.11
       │ │ ├─┬ accepts@1.1.4
       │ │ │ └── negotiator@0.4.9
       │ │ ├── base64id@0.1.0
       │ │ ├─┬ debug@2.2.0
       │ │ │ └── ms@0.7.1
       │ │ ├─┬ engine.io-parser@1.2.4
       │ │ │ ├── after@0.8.1
       │ │ │ ├── arraybuffer.slice@0.0.6
       │ │ │ ├── base64-arraybuffer@0.1.2
       │ │ │ ├── blob@0.0.4
       │ │ │ ├── has-binary@0.1.6
       │ │ │ └── utf8@2.1.0
       │ │ └─┬ ws@1.1.0
       │ │   ├── options@0.0.6
       │ │   └── ultron@1.0.2
       │ ├─┬ has-binary@0.1.7
       │ │ └── isarray@0.0.1
       │ ├─┬ socket.io-adapter@0.4.0
       │ │ ├─┬ debug@2.2.0
       │ │ │ └── ms@0.7.1
       │ │ └─┬ socket.io-parser@2.2.2
       │ │   ├── debug@0.7.4
       │ │   └── json3@3.2.6
       │ ├─┬ socket.io-client@1.4.8
       │ │ ├── backo2@1.0.2
       │ │ ├── component-bind@1.0.0
       │ │ ├── component-emitter@1.2.0
       │ │ ├─┬ debug@2.2.0
       │ │ │ └── ms@0.7.1
       │ │ ├─┬ engine.io-client@1.6.11
       │ │ │ ├── component-inherit@0.0.3
       │ │ │ ├─┬ debug@2.2.0
       │ │ │ │ └── ms@0.7.1
       │ │ │ ├── has-cors@1.1.0
       │ │ │ ├── parsejson@0.0.1
       │ │ │ ├── parseqs@0.0.2
       │ │ │ ├── ws@1.0.1
       │ │ │ ├── xmlhttprequest-ssl@1.5.1
       │ │ │ └── yeast@0.1.2
       │ │ ├── indexof@0.0.1
       │ │ ├── object-component@0.0.3
       │ │ ├─┬ parseuri@0.0.4
       │ │ │ └─┬ better-assert@1.0.2
       │ │ │   └── callsite@1.0.0
       │ │ └── to-array@0.1.4
       │ └─┬ socket.io-parser@2.2.6
       │   ├── benchmark@1.0.0
       │   ├─┬ debug@2.2.0
       │   │ └── ms@0.7.1
       │   └── json3@3.3.2
       ├─┬ superagent@0.21.0
       │ ├── component-emitter@1.1.2
       │ ├── cookiejar@2.0.1
       │ ├── extend@1.2.1
       │ ├─┬ form-data@0.1.3
       │ │ ├── async@0.9.2
       │ │ └─┬ combined-stream@0.0.7
       │ │   └── delayed-stream@0.0.5
       │ ├── formidable@1.0.14
       │ ├── methods@1.0.1
       │ ├── mime@1.2.11
       │ ├── qs@1.2.0
       │ ├─┬ readable-stream@1.0.27-1
       │ │ ├── core-util-is@1.0.2
       │ │ ├── inherits@2.0.1
       │ │ └── string_decoder@0.10.31
       │ └── reduce-component@1.0.1
       └─┬ vd@0.1.0
       ├── array-flatten@1.0.0
       ├── vcss@0.1.1
       └── void-elements@1.0.0
       
-----> Caching build
       Clearing previous node cache
       Saving 2 cacheDirectories (default):
       - node_modules
       - bower_components (nothing to cache)
-----> Build succeeded!
       ├── args@1.3.0
       ├── babel-polyfill@6.3.14
       ├── body-parser@1.10.2
       ├── debug@2.1.1
       ├── email-regex@1.0.0
       ├── express@4.11.0
       ├── hostenv@1.0.1
       ├── opentype.js@0.4.4
       ├── socket.io@1.4.8
       ├── superagent@0.21.0
       └── vd@0.1.0
       
-----> Discovering process types
       Procfile declares types -> web
-----> Compressing...
       Done: 13.7M
-----> Launching...
       Released v4
       https://chirimen-oh.herokuapp.com/ deployed to Heroku```

gurezo, 2016-07-20:
最新のslackのAPIトークンで、再構築を後日試みる
アカウントは復活済み

gurezo, 2016-08-02:
[公開Slack用 自動招待フォーム by GAS](http://qiita.com/atusi/items/c2669add8038c43efff3)作成出来たので、close

closed 2016-08-02 by gurezo