chirimen-oh/any-issues #20: github名も統一しますか？

gurezo, 2016-04-24:
## slackでの投稿より
### @naobsd さんから 下記指摘が。
- 「githubはchirimen-ohじゃないんですね...」

### github名も統一しますか？
- @MSakamaki さん、github名って、chirimen-org-> chirimen-ohって出来るんでしたっけ？  
Organizationsって、やった事ないので分からないのですが・・・。
- 4/18 に@dynamis さんと @dadaa さんで、[http://chirimen.org/ ](http://chirimen.org/ ) で  
  まとまったと認識してるのですが・・・・。  
  話を蒸し返してる感があるので、皆様の意見を下さいませ。

MSakamaki, 2016-04-25:
chirimen-ohは文字の長さでohと記憶してたので、ちゃんと名前がつけれる所はChirimenOpenhardにしたほうが良いと考えてます
ドメインとかも含めて一度話した方がいいかもですね。

gurezo, 2016-04-25:
### 名前も微妙に違いが・・・・。
- CHIRIMEN Open Hardware（投票時）
- CHIRIMEN Open Hard（うろ覚え）

---- 

**※どこかで議論しましょう！**
文字数の制約もあるますね・・・・。

satakagi, 2016-04-27:
CHIRIMEN Open Hardは時々MozOpenHardとの混同から使われていることがあるのかもしれませんが、CHIRIMEN Open Hardwareが正ですよね？

dadaa, 2016-04-27:
CHIRIMEN Open Hardware が正ですー

satakagi, 2016-04-27:
それと、ドメイン名は、
>4/18 に@dynamis さんと @dadaa さんで、http://chirimen.org/  でまとまった
で合意できていると思っています。今別の名前に変えることを議論するのは確かに時機を逸していると思います。
名称がCHIRIMEN Open Hardwareでもそのドメイン名がchirimen.orgというのは　尾のすごく違和感があるとも言えない（特にOpen Hardwareの部分は固有名詞とは見えないため）ので、CHIRIMENがドメイン名として取得できているならば、それでも悪くはないと思いますがいかがでしょう。

gurezo, 2016-04-27:
では、まとめとして
正式名称：CHIRIMEN Open Hardware
github名：http://chirimen.org/

上記で問題なければ、この議論をcloseしようとお思います。
如何でしょう？

satakagi, 2016-04-27:
私は賛成です^^

gurezo, 2016-04-27:
@MSakamaki さん、
@dadaa さん、
@naokisekiguchi さん、
@naobsd さん、如何でしょうか？？？

dadaa, 2016-04-27:
ひとまず現状を：
ホームページURL: http://chirimen.org/
Slack: chirimen_oh
twitter: chirimen_oh
GitHub: chirimen-org
(これから Facebook url: chirimen_oh?)

gurezo, 2016-04-27:
そのまとめで良いと思います。

dadaa, 2016-04-27:
しかし，たしかにこうならべると，chirimen_oh の方が統一感はありますね〜

gurezo, 2016-04-27:
それなので、issueを立てちゃいました。
http://chirimen-oh.org/

なまじ、上記ドメイン名が取得できる状況だけに・・・・。
本当に混ぜ返し／蒸し返して、すみません。

dadaa, 2016-04-27:
なんか，ドメインは chirimen.org がステキな感じしますけど，他利用サービスは chirimen_oh 統一が気持ちよい感じがしました

gurezo, 2016-04-27:
自分が気になるのは、仮に http://chirimen-oh.org/ 変更したとして、
今皆さんが作業されている労力が、また発生する事が容易に想像出来るので・・・。

dadaa, 2016-04-27:
あ，いえ，URL は http://chirimen.org で，GitHub のグループを chirimen_oh にするって感じです．

dadaa, 2016-04-27:
すると，
ホームページ: http://chirimen.org/
Slack: chirimen_oh
twitter: chirimen_oh
GitHub: chirimen_oh
Facebook url: chirimen_oh

dadaa, 2016-04-27:
どうかしら？

gurezo, 2016-04-27:
あ、そうゆう意味ですね！
理解しました！
で、あればこれで議論をまとめましょうか？
@MSakamaki さん、
@satakagi  さん、
@naokisekiguchi さん、
@naobsd さん、宜しいでしょうか？？？

gurezo, 2016-04-27:
@dadaa さんのまとめで、closeしようと思います。

ホームページ: http://chirimen.org/
Slack: chirimen_oh
twitter: chirimen_oh
GitHub: chirimen_oh
Facebook url: chirimen_oh

dadaa, 2016-04-27:
みなさん，どうでしょう？

gurezo, 2016-04-27:
では、異論なさそうなので、close します

closed 2016-04-27 by gurezo

naobsd, 2016-04-29:
出遅れてすみません、私は元々githubの名前が気になっただけでchirimen.orgを変える話はしてませんでしたので、結論に基本的には異論はないです。
強いて一つ言うならば、ホスト名にunderscoreは使えないので、サービスによってはunderscoreが使えない場面が出てくる可能性があるかもと思います。