chirimen-oh/any-issues #23: https://github.com/chirimen-orgのリポジトリの説明がほとんどない　このissuesのリポジトリの説明もほとんどない

satakagi, 2016-04-26:


dadaa, 2016-04-26:
any-issues の説明に関する issue は，以下に登録してあります．
https://github.com/chirimen-org/any-issues/issues/7

naokisekiguchi, 2016-04-26:
説明文が解決されたらチェックをつけていきましょう。

- [ ] device-chirimen
- [ ] linux-rockchip
- [ ] platform_system_core
- [ ] device-rockchip-rksdk 
- [ ] gecko-dev
- [ ] B2G
- [ ] b2g-manifest
- [ ] examples
- [ ] chirimen-org.github.io
- [ ] any-issues
- [ ] CHIRIMEN-tools
- [ ] b2g-patches
- [ ] gaia 
- [ ] i2c-tools
- [ ] u-boot-rockchip

naokisekiguchi, 2016-04-26:
私がわかる範囲で記載してみました。?がついているのはよくわからない部分です。その他も間違いがあるかもしれませんが修正してもらえると。

- device-chirimen
??WebIDEにCHIRIMENと認識させるための変更とか？

- linux-rockchip
??CHIRIMENのチップ(rk3066)用のカーネルを作るのに必要？u-boot-rockchipとの違いはよくわからないです。

- platform_system_core
??

- device-rockchip-rksdk 
??I2CやGPIOを操作するためのpermissionの設定など?

- gecko-dev
B2Gのgecko層をCHIRIMEN用にカスタマイズしている。webGPIO/I2C APIの実装もここに組み込まれる。

- B2G
B2Gをビルドする際の大元となるレポジトリ。CHIRIMEN用のビルドコンフィグなどが追加されている。イメージのビルドをしたい場合、このレポジトリをcloneする。

- b2g-manifest
各モジュールのレポジトリを指定している。CHIRIMENビルド用のレポジトリの指定が追記されている。

- examples
WebGPIO/I2C APIを使用したアプリのサンプル集。CHIRIMENで動作確認の取れたデバイス（センサー、アクチュエーター）の基本動作用プログラム集。

- chirimen-org.github.io
CHIRIMENホームページのソースコード

- any-issues
CHIRIMEN開発全体に関わる課題を管理するためのissues用レポジトリ

- CHIRIMEN-tools
CHIRIMEN開発に役立つツール集。Windows用ドライバ、イメージ作成ツールやFritzing（回路図作成ツール）用のCHIRIMEN素材など。

- b2g-patches
B2GをCHIRIMEN用にビルドして動作させるためのpatch集。いずれはgecko-devにマージしてこのレポジトリをなくすことが求められる。

- gaia 
B2Gのgaia層。現在はCHIRIMEN用のカスタマイズは入っていない？CHIRIME起動時のアプリやホーム画面を作っていく場合はこのレポジトリを編集することになる

- i2c-tools
i2c-toolsというコマンドライン上でI2Cの操作を行うためのプログラム。shell上でのI2C動作確認用にCHIRIMENに組み込むためのレポジトリ。

- u-boot-rockchip
??CHIRIMENのチップ(rk3066)用のカーネルを作るのに必要？

MSakamaki, 2016-04-26:
とりあえず、examplesとホームページだけ追加したもののブランチを切りました。(`issues/#7`)
こちらに上にあるリストを追加して行く形でいいでしょうか？
https://github.com/chirimen-org/any-issues/pull/29

kohichi000000, 2016-04-26:
私はいいと思いますが、
・今後、新たにリポジトリが追加されたら、必ずこのreadmeに説明が加わるということでいいでしょうか？
　（もしそうであれば、外部からgithubへのリンクを書く際に、このreadmeを参照してほしいと書くとか）
・関係のあるものはできるだけまとめて書いたほうがいいと思いますがいかがでしょうね（順番の問題）。

naokisekiguchi, 2016-04-26:
CHIRIMEN開発に関わる全てのissueはany-issuesレポジトリで管理すると思っていたのですが、READMEを見ると、各レポジトリに閉じた課題だと明らかな場合は各レポジトリでissueを挙げる感じでしょうか？

MSakamaki, 2016-04-26:
> ・今後、新たにリポジトリが追加されたら、必ず上記readmeに説明が加わるということでいいでしょうか？
>　（もしそうであれば、外部からgithubへのリンクを書く際に、上記readmeを参照してほしいと書けば、いろんな混乱を防げるかも）

こちらの文言を追加してみました。
https://github.com/chirimen-org/any-issues/tree/issues/%237

> CHIRIMEN開発に関わる全てのissueはany-issuesレポジトリで管理すると思っていたのですが、READMEを見ると、各レポジトリに閉じた課題だと明らかな場合は各レポジトリでissueを挙げる感じでしょうか？

すいません、ここら辺の文言は仮でした。
リポジトリ毎の議題が混じっても面倒なので、以下どっちかになるかなと思っています。

 1. リポジトリ毎にissues毎に管理する
 2. any-issues内でタグ管理する。
   + `/chirimen-org/any-issues/issues/new?labels=[ラベル名]`でページ遷移させると、既にラベルがついた状態で新規issues作成ページへジャンプさせる事ができます。

ここらへん決まっていましたっけ？

naokisekiguchi, 2016-04-26:
明確には決めていなかった気がします。
個人的には
2.any-issues内でタグ管理する。
が好みです。

kohichi000000, 2016-04-26:
naokisekiguchiさんのコメントを参考に、自分が書いた通りに分類してみました。大雑把に分けるとこんな感じですかね。太字の部分が補強したい点です（疑問点）。
---

#### CHIRIMENに関する議論（掲示板）
##### any-issues
* CHIRIMEN全体に関わるissueを議論するための掲示板。下記レポジトリの説明も含まれる(readme)。

#### CHIRIMENを利用した開発、CHIRIMENに関する情報に関するリポジトリ
##### examples
* CHIRIMENで動作確認の取れたデバイス（センサー、アクチュエーター）をWebGPIO/I2C APIを使用し
動作させるための基本プログラム集。
デバイスの利用法がfabbleで説明され、そのためのプログラムが本レポジトリに収納される。

##### chirimen-org.github.io
* CHIRIMENホームページのソースコード。
markdown記法でホームページを書いて掲載するためにはjekyllによる変換が必要。

##### CHIRIMEN-tools
* CHIRIMENの開発、ドキュメント作成に役立つツール集。
Windows用ドライバ、イメージ作成ツールやFritzing（回路図作成ツール）用のCHIRIMEN素材など。

#### CHIRIMEN用ビルドを作るために必要となるモジュールのレポジトリ
##### B2G
* B2Gをビルドする際の大元となるレポジトリ。
CHIRIMEN用ビルドのためのconfigなどが追加されている。
CHIRIMENに焼くイメージをビルドをしたい場合、このレポジトリをcloneする。

##### b2g-manifest
* 各モジュールのレポジトリを指定。
CHIRIMENビルド用のレポジトリ指定が追記されている。

##### gaia 
* B2Gのgaia層に相当。
CHIRIME起動時のアプリやホーム画面を作る場合はこのレポジトリを編集する。
**現在はCHIRIMEN用のカスタマイズは入っていないと思われるため特別に必要か？**

##### b2g-patches
* B2GをCHIRIMEN用にビルドして動作させるためのpatch集。
いずれはgecko-devにマージしてこのレポジトリをなくすことが求められる。

#####  gecko-dev
* B2Gのgecko層をCHIRIMEN用にカスタマイズしたもの。webGPIO/I2C APIの実装もここに組み込まれる。
b2g-manifestから参照されている。

#####  i2c-tools
* コマンドライン上でI2Cの操作を行うためのLinuxプログラム。
shell上でのI2C動作確認用にCHIRIMENに組み込まれる。
b2g-manifestから参照されている。

#####  device-chirimen
* **WebIDEにCHIRIMENと認識させるための変更とか？**
b2g-manifestから参照されている。

#####  device-rockchip-rksdk 
* **I2CやGPIOを操作するためのpermissionの設定など?**
b2g-manifestから参照されている。

#####  u-boot-rockchip
* **CHIRIMENのチップ(rk3066)用のカーネルを作るのに必要？**
b2g-manifestから参照されている。

#####  linux-rockchip
* **CHIRIMENのチップ(rk3066)用のカーネルを作るのに必要？**
**u-boot-rockchipとの違いを書きたいが…**
b2g-manifestから参照されている。

#####  platform_system_core
* **GPIOに対し正しくpermissionを設定するための回避策として必要。**
b2g-manifestから参照されている。

MSakamaki, 2016-04-26:
一旦上のmarkdownをReadmeに転記した物を以下ブランチに適応しておきます。
https://github.com/chirimen-org/any-issues/tree/issues/%237

dadaa, 2016-04-26:
b2g-manifest : B2G から参照?(というのがいいのかはちょっとわからないです)されます．
// B2G の confing.sh で b2g-maniest が git clone されます．

gaia, b2g-patches, gecko-dev, i2c-tools, device-chirimen, device-rockchip-rksdk, u-boot-rockchip, linux-rockchip, platform_system_core は b2g-manifest から参照されます．（上記 config.sh で結果的に B2G の下にこれらリポジトリが clone されます）

kohichi000000, 2016-04-27:
ありがとうございます。こんな感じですかね？

##### b2g-manifest: 
* B2Gのconfig.shで以下の各リポジトリ（他、外部のリポジトリも含まれる）をcloneするための参照情報。以下のリポジトリがb2g-manifest から参照され、結果として、B2Gのconfig.sh でこれらのリポジトリが B2G の下にclone されます。

kohichi000000, 2016-04-28:
上記更新しました。

MSakamaki, 2016-04-28:
examplesにfabbleへリンク付けた方が良いでしょうか？

dadaa, 2016-04-28:
あ，賛成です！ > examples に fabble へのリンク

naokisekiguchi, 2016-04-28:
Lチカとbuttonはfabble修正したのでリンク貼っときました。

naobsd, 2016-04-29:
b2g関連のリポジトリ、とりあえず一つ例にあげますが

> platform_system_core
> 
> GPIOに対し正しくpermissionを設定するための回避策として必要。 b2g-manifestから参照されている。

たとえばこれは、CHIRIMEN用ビルドを作るために必要となるモジュールのレポジトリ・b2g-manifestから参照されている、で十分であり、GPIOのpermission云々はたまたま今そういうパッチが入っているだけであって、リポジトリの説明としては適切ではないと思います。

もちろん、現状で動作するものをビルドする為にはこのリポジトリに入っているパッチが必要です、という情報がどこかにないといけないのですが、それはたぶんissueとして存在すべきものという認識です。

b2g関連のリポジトリは、頭にb2g-でもつけて、CHIRIMEN用ビルドを作るために必要となるモジュールのレポジトリです、で済ませる方がよいのではと思いますが、どうでしょうか？

kohichi000000, 2016-04-29:
「b2g関連のリポジトリは、頭にb2g-でもつけて、CHIRIMEN用ビルドを作るために必要となるモジュールのレポジトリです」でいいと思います。初めてきた人が見て何なのかがわかればそれでいいと思います。

gurezo, 2016-07-03:
### repository：meeting を追加予定
* 目的：変更履歴を残す事と slack との通知連動の為
* 現行：any-issue の wiki に記載されている内容を  
  ミーティング毎に記載予定

MSakamaki, 2016-07-08:
issues/#7 ブランチマージしちゃっていいですか？

👍  が３つ集まればマージします。
👎 が１個でもあればちょっと待ちます。

MSakamaki, 2016-07-19:
あとひとり来ないけれ２人も３人も同じよということで、#7 マージしちゃいますね。

gurezo, 2016-08-07:
説明も一通りできていると様なので、close

closed 2016-08-07 by gurezo