chirimen-oh/any-issues #32: Rearrange slack channel

naokisekiguchi, 2016-04-29:
slackのchannelがわかりにくいので整理して説明を記載

naobsd, 2016-04-29:
#eventsはイベント関連の情報が流れる、でいいとして、#scheduleは？
#webは？projectのwebどーしましょって話はissuesでやってますよね？
#devと#portingの使い分けは？たぶんWebAPIを使うアプリを作る話と、CHIRIMENの上でB2Gを動かす話は、分けた方がよいとは思うのですが(WebAPIを実装する話はどっちだろう...)

kohichi000000, 2016-04-29:
できれば、ホームページやgithubのリポジトリとの整合もわかるようにしたい

dadaa, 2016-04-30:
チャネルの説明を加える

naobsd, 2016-05-15:
(#33 に貼ってあったのをこっちにも)
https://docs.google.com/spreadsheets/d/1ibBG4Ee_COF6rRQWu1-qVpC9zTGwcScTRnXU4xjAxHU/edit?usp=sharing

naobsd, 2016-05-15:
#scheduleと#webは削除していただきました

naokisekiguchi, 2016-06-21:
不足していた(?)webapiとhardwareチャンネルを加えて、各チャンネルのpurposeに説明を追加しました。どなたか確認、必要なら修正してくださいー。

上のgoogle docみると#devはアプリ開発とかでしたね。
#dev は#dev-appとかの名前にしたほうがわかりやすい気がしますが、いかがでしょう？
（チャンネル作成者か管理者しか名前は変更できない）

naobsd, 2016-07-08:
issueのラベル、platformをportingに変更しちゃいました(slackの#portingにあわせました)

gurezo, 2016-08-07:
説明文を @naokisekiguchi さんに書いて頂いたのと #105 で対応したので、close

closed 2016-08-07 by gurezo