chirimen-oh/any-issues #36: NodeBots Dayについて

ega1979, 2016-04-30:
7/30にMozilla Japanさんで「NodeBots Day」を開催したいと思います。
J5などJavaScriptを使って作って騒ごうというハッカソン的＆もくもく会な感じにしたいですが、このイベントにchirimenもJoinできないかなと考えていますが、どうでしょう？

MSakamaki, 2016-05-06:
@ega1979 さん
これって申込やもろもろの期日って何時迄ですか？
余裕があるなら次回MTGのテーマにするのが良いかなと思います。

ega1979, 2016-05-08:
@MSakamaki さん
期日は特に決めてないです。後乗りでもOKなので、告知後でもぜんぜんウェルカムです。

ちなみに今日IIJも使うことができると連絡があったので（135人OKの会場なので）。ハンダ行けるか確認します。
ハンダOKなら規模をどの程度するか次第ですね。個人的には50人以上はキツいかなと。

n0bisuke, 2016-06-20:
Issueがあったんですね！

MSakamaki, 2016-07-07:
NodeBots Dayですが、とりあえず以下の内容になりそう。

基本こっちの参加スタンスは「CHIRIMENのタッチ＆トライをしてもらう」

 1. もくもく会ベース
 2. CHIRIMENチームはブースを提供してもらえるので。CHIRIMENハードウェアを提供して、タッチ＆トライをしてもらう。
 3. センサー類やハンダなどの提供は、各コミュニティや参加者自前

「確認中」
どれだけのサイズのエリアを提供してもらえるのか・・・

gurezo, 2016-08-07:
イベントが終了したので、close

closed 2016-08-07 by gurezo