chirimen-oh/any-issues #37: Google Drive の活用法

gurezo, 2016-05-06:
#### chirimen.open.hardware@gmail.com のGoogle Drive の活用（15GB）
##### 用途
1. イベント用資料
1. イベント用配布物（zip？）
1. バックアップ？（安定版のイメージとか）
1. その他（？？？？）

##### 権限の範囲
1. 編集可：CHIRIMEN コントリビューター
1. 編集可：CHIRIMEN チューター
1. 閲覧のみ：一般？

-----

どうやって範囲決めしたら良いか、アドバイスお願いします

naobsd, 2016-05-15:
Googleカレンダー活用できますかねってちょっと思いついたのでとりあえずここに書いときます

MSakamaki, 2016-05-16:
googleカレンダーは良いですね。 👍

gurezo, 2016-05-16:
## CHIRIMEN イベントカレンダー作成
##### Google カレンダーで、CHIRIMEN Open Hardware に関連するイベントを公開しました。
###### 適用のやり方は、下記の通り
1. Google カレンダーで、開く
1. 他のカレンダーをクリック
1. 友達のカレンダーの追加
1. メールアドレス：chirimen.open.hardware@gmail.com を入力
1. 追加ボタンをクリック
1. 完了

naobsd, 2016-08-09:
chirimen.open.hardware@gmail.com を名刺に記載しているとの事
(どなたがメールを読み書きできるのかは把握していません)

gurezo, 2016-08-09:
現時点では、 @gurezo と @dadaa さん

naobsd, 2016-08-09:
名刺って他の方も作られたのですよね？

gurezo, 2016-08-09:
はい、一律同じアドレスになっています。