chirimen-oh/any-issues #38: maker faire 2016

MSakamaki, 2016-05-11:
## 出展内容

 + 展示 (デモ)
 + ハンズオン

gurezo, 2016-07-03:
[上記タイムスケジュール（暫定）](https://drive.google.com/open?id=18LtwfeaJXBBv7pB382Lsuvljbilr_kkLGk-EdDuuiug)

gurezo, 2016-07-03:
### 出展物予定
* CHIRIMEN Open Hardware コミュニティ枠
 * [ (8/6-7) Maker Faire Tokyo 2016 #38 ](https://is.gd/loOpvE)
 * タッチ＆トライ・イベント
  * [satakagi](https://github.com/satakagi) さんが2つ作っている
    * プラネタリウム（ケータイ星座早見板＋LED）製作キット
    * 着陸ゲーム
* KDDIブース 枠
  * リケジョ（仮
  * ディスカッションセッション
    * [HAKUTOのミッション](https://team-hakuto.jp/mission/index.html)についてCHIRIMENで何かできないかガチ議論。CHIRIMENが宇宙に！
  * パネルセッション（坂井先生枠 => ご相談
    * 誰が出たら面白い？：
      * [HAKUTO](http://team-hakuto.jp/index.html)さん
      * [オライリー](http://www.oreilly.co.jp/index.shtml)さんにも依頼
      * スイッチサイエンスさん
* [HAKUTO](http://team-hakuto.jp/index.html) 枠
  * 何かとコラボ企画：

gurezo, 2016-08-07:
イベントが終了したので、close

closed 2016-08-07 by gurezo