chirimen-oh/any-issues #39: CHIRIMENステッカー

gurezo, 2016-05-11:
#36 や #38 とかで、アピール用にチラシ他
ステッカーを用意した方が良いかと思います。

サイト上で、ちょっと調べてみました。
[シール印刷の専門サイト - weblabels.net‎](https://is.gd/K2GlgC)
[シール印刷・ステッカー印刷・ラベル印刷](http://www.graphic.jp/lineup/ondemand/seal_sticker.php)

他に良い情報があれば、そちらでやりましょう。

課題
- [ ] 予算
- [ ] デザイン

kazygumi, 2016-05-13:
デザインは自分で良ければやりますよ

kohichi000000, 2016-05-16:
#40が一番最初に来るイベントなので、ちょっとばかし急ぎたいです（せめてチラシだけでも）。課題は同じ。

gurezo, 2016-05-16:
もし、急ぐのであれば、下記提案です。
**[CHIRIMENアイコン](https://drive.google.com/open?id=0B4V6iMhRJyKOZFFrWDZQc1JaSTA)**
上記ロゴステッカーの画像で、入稿する。

kohichi000000, 2016-05-16:
チラシについては、昨年から使っていたものを使うのがいいと思っているのですが、URLなどいくつか文言変える必要があるので、後ほどissueたてます。

kazygumi, 2016-05-16:
6/8であれば、チラシもシールも来週中に入稿すれば間に合いますね。
自分の方でデザインと見積を進めましょうか？
お金は割り勘ですか？

gurezo, 2016-05-16:
自分は費用に協力する予定です。
割り勘するにしても誰が参加してくれるでしょ（苦笑

デザインと見積もりお願いします！

kohichi000000, 2016-05-16:
gurezoさん、kazygumiさんありがとうございます。
はい、そんな感じです。来週中にはfixしたいです。
チラシについて文案等作っているので、議論できるように別issue立てます。

MSakamaki, 2016-05-19:
>お金は割り勘ですか？

酒巻さんも手を挙げさせてもらいます。

kazygumi, 2016-05-20:
金額はおそらく下記のような感じになります。

■チラシ
両面 A4 500部 **3,300円**（200部から500部までほぼ同じ値段）

■円形シール
直径5cm 300枚 1個ずつシートをカット **10,414円**

kohichi000000, 2016-05-20:
ありがとうございます
チラシですが
A5、カラー+白黒、110g、2000枚、光沢
だとどのくらいでしょ？
わたしの認識だと4200円くらい。

kazygumi, 2016-05-20:
そうですね。
もし来週早々にデータが完成すれば4,200円でできると思います。
遅くなったとしても5,000円前後です。

ただ2,000枚となると配りきれないと思いますので、
イベント後の置き場所を考えなくてはならない気がします。
Mozillaさんとか置かせてくれたりしますかね。

kohichi000000, 2016-05-20:
基本はうちにおいていいですが、モジラさんでもおさわりとかしたとき用にある程度置かせていただけるとうれしいです。
多分maker faireまでで1000位配ると思うので、そんくらいあってもいいですよね？

gurezo, 2016-05-20:
良いと思います。
支払いとかは、一度梶井さんにお願いしちゃって良いですか？
来週か再来週割り勘分はモジラオフィスに預ければ良いですよね？
一人当たり1,500円〜2,000円？

gurezo, 2016-05-20:
チラシ＋シールだから、合計15,000円は見なくてはですね
一人当たり4,000円〜5,000円ですね
勘違いしてました(^^;

kohichi000000, 2016-05-25:
CHIRIMENロゴのベクターデータってありますか？？
ロゴとして掲載したいので。
（展示しようとするといわれそう）

kohichi000000, 2016-05-25:
やはりなんですが
Interopの事務局から、ロゴのベクターデータを明日(5/26)の10時くらいまでにほしいといわれてしまいました。
可能でしょうか？？
丸い方だけで大丈夫です。

kazygumi, 2016-05-25:
渡せるロゴのベクターデータを作ってみました。

[CHIRIMEN-VI.pdf](https://github.com/chirimen-oh/any-issues/files/281636/CHIRIMEN-VI.pdf)

どのような使われ方をするのかわからなかったので、
念のため「CHIRIMEN」のタイポも含めたパターンも作ってみました。
「CHIRIMEN」の文字が無いような場所に掲載されるのであれば、
C.かD.を使っていただければと思います。

kazygumi, 2016-05-25:
念のため丸いロゴだけのものもアップしておきます。

[CHIRIMEN-logo-circle.pdf](https://github.com/chirimen-oh/any-issues/files/281809/CHIRIMEN-logo-circle.pdf)

kohichi000000, 2016-05-25:
ありがとうございます

kohichi000000, 2016-05-25:
コメントが足りませんでした。各ブースにおいて団体名、テーマを掲げるんですが、その横にロゴが載る形です。つまり、タイポは無くても大丈夫なので、A案を使いたいと思います。
A案のベクタ画像があればどっかに上げてもらえると嬉しいです。（ジャギーが出るのを防ぐため）

kohichi000000, 2016-05-25:
チラシも固まったと思うので、今日明日くらいでステッカーとともに入構作業に入りましょうかね。
@gurezo さんがおっしゃるとおり、 @kazygumi さんにお願いしてしまっていいでしょうか？
（お忙しいようであれば私の方でも可能です）

まとめると、
* スペック：
 * チラシ： #42 参照。A5、カラー+白黒、110g、2000枚、光沢
 * 円形（上記参照）。直径5cm 300枚 1個ずつシートをカット
* 費用： 
 * 可能であれば @kazygumi さんの方で立て替えていただけるとうれしいです 
 * 今のところは手をあげてくださった方で割り勘
 * 場合によっては工藤さんがfacebookにあげていたファンドを使う可能性等もある(New!)
* 送付先：　
 * 納入が6/7でよければInterop会場直接送付でも可能(その場合は送付先住所を伝えます)。
 * その前に一度見ておきたいのであれば別途検討（最終の置き場は弊社(KDDI)でもいいです。モジラさんともご相談。）

いかがでしょうか？

naokisekiguchi, 2016-05-26:
今まで議論に気づいてませんでした。
#42 にコメントしましたがチラシに載せるプログラムを2行くらい削減できるのでご確認いただけますか？

kazygumi, 2016-05-26:
@kohichi000000 さん >
> A案のベクタ画像があればどっかに上げてもらえると嬉しいです。（ジャギーが出るのを防ぐため）

下記のファイルはベクタ画像ですのでそのままご利用いただけます。
[CHIRIMEN-logo-circle.pdf](https://github.com/chirimen-oh/any-issues/files/281809/CHIRIMEN-logo-circle.pdf)

入稿作業、入金に関して了解しました。こちらで対応します。
送付先はInterop会場としようと思いますので
送付先をお教えいただいてもよろしいでしょうか。

@naokisekiguchi さん >
了解しました。確認します。

kohichi000000, 2016-05-26:
> 送付先はInterop会場としようと思いますので
> 送付先をお教えいただいてもよろしいでしょうか。

ありがとうございます。
個人情報を書く必要があるので、後ほど @kazygumi さんにメールします。

kohichi000000, 2016-05-26:
それぞれ残ったら、コミュニティの財産としてモジラジャパンさんにおいていただくことになりました。

kazygumi, 2016-05-26:
チラシの入稿が完了しました。
業者はラクスルさんで5,832円でした。

kazygumi, 2016-05-26:
シールも入稿done。
シール直送便さんで9,075円でした。

gurezo, 2016-05-26:
@kazygumi さん、チラシ、シールデザインから入稿まで有難うございました。

kazygumi, 2016-05-27:
すみません！
ステッカーの件ですが、
誤ってシール素材を選んでしまっていたので、ステッカー素材に変更します。
そのため金額が10,683円に変更となりました。

なので合計金額は下記のとおりとなります。
チラシ　　　5,832円
ステッカー　10,683円
----------------------------------
合計　　　　16,515円

kazygumi, 2016-05-27:
割り勘に参加頂けるメンバーは、
@gurezo 
@kohichi000000 
@MSakamaki 
@kazygumi
の4名であっていますでしょうか。

1人4,130円でどうでしょう。

kohichi000000, 2016-05-27:
いまのところそうですが、もちっとおうぼしてみましょか？

naokisekiguchi, 2016-05-27:
私も参加しますよ。
CHIRIMEN dayから1ヶ月たつのでマンスリーミーティングをやって人を集めましょうか。

gurezo, 2016-05-27:
どこかで、集金しなきゃですね。
自分は、カンパオッケーです。

dadaa, 2016-05-27:
僕も！

kazygumi, 2016-05-27:
またまたすみません。
ラクスルさんが日時指定できないようなので、
一旦弊社で受け取ってから日時指定で送るようにします。
なので追加で宅急便代が1000円程度かかる予定です。
（A1パネルも作ることになったので、うまくいけばまとめて送れるかも）

kohichi000000, 2016-05-27:
パネルの印刷どうします？

kazygumi, 2016-05-27:
キンコーズでよければ会社の近所にあるのでこちらで印刷できますよ。
A1一枚8000円くらいだったと思います。

kohichi000000, 2016-05-27:
そんくらいかかりますよね。
お金かけたくないので、私の方でW3C（というか大学）でやろうと思うんですが、いいですかね？
W3C主催のイベントのためだし、許されるかと（一応確認中）

kazygumi, 2016-05-27:
了解しました！
大学側で可能であれば、デザインが出来しだい
デザインデータをGoogle Driveあたりで共有するようにします。

gurezo, 2016-06-20:
Interop Tokyo 2016にて、作成・配布等々が出来たので、closesします

closed 2016-06-20 by gurezo