chirimen-oh/any-issues #41: (6/8-10) Interop Tokyo 2016

naobsd, 2016-05-15:
http://exhibitor.f2ff.jp/2016/exhibition_form/companylist.html 7P33

kohichi000000, 2016-05-16:
## 出展内容
- 展示（デモ）
 - 詳細は別途検討（基本はアリもの）

## 枠組み
- W3Cブース内にコミュニティ枠というのができます。その中で出す予定。

## 懸案事項
- 展示員（多分１－２名）
 - 参加したい人（できそうな人）は適宜手をあげてください。6/8-10の一部で全然問題ないです。平日ですので無理のない範囲で。
- チラシ（刷新したもの）

gurezo, 2016-05-16:
## 展示員スケジュールフォーマット
- [ ] 06/08 AM
- [ ] 06/08 PM
- [ ] 06/09 AM
- [ ] 06/09 PM
- [ ] 06/10 AM
- [ ] 06/10 PM

gurezo, 2016-05-16:
## きぐれ日程
#### 現時点なら、06/09(木)終日お手伝いできます。
- [ ] 06/08(水) AM
- [ ] 06/08(水) PM
- [X] 06/09(木) AM
- [X] 06/09(木) PM
- [ ] 06/10(金) AM
- [ ] 06/10(金) PM

kohichi000000, 2016-05-19:
今回のイベントはW3C/Asiaの拠点が慶應になってから２０周年であることを祝して行われるのですが、
6/9夜（19:20-）に開催されるレセプションに本コミュニティの方もご参加いただけることとなりました。8888
参加したいという方は何らかの手段で高木までご連絡ください。参加登録方法をお伝えします。
https://www.w3.org/20/Asia/Japan/Overview.html

kohichi000000, 2016-05-24:
ブースに掲げるテーマとロゴですが、5/25（水）正午を〆切としたいと思います。
以下の案で問題あれば突っ込んでください。何もなければこのまま確定します（提出します）

ロゴ：　
案（丸い方）： https://drive.google.com/file/d/0B4V6iMhRJyKOZFFrWDZQc1JaSTA/view

テーマ（40文字以内）：
案：　デバイスをブラウザで操作するためのボード・APIの開発をオープンコミュニティで推進

kohichi000000, 2016-05-24:
## 高木日程：責任者ということで一応ずっといます（ひょっとするとhtml5jの方に顔出しているかも）
- [x] 06/08(水) AM
- [x] 06/08(水) PM
- [x] 06/09(木) AM
- [x] 06/09(木) PM
- [x] 06/10(金) AM
- [x] 06/10(金) PM

kohichi000000, 2016-05-25:
これで確定します！

>  ロゴ：　
> 案（丸い方）： https://drive.google.com/file/d/0B4V6iMhRJyKOZFFrWDZQc1JaSTA/view
> テーマ（40文字以内）：
> 案：　デバイスをブラウザで操作するためのボード・APIの開発をオープンコミュニティで推進

kohichi000000, 2016-05-26:
明日出展者証を確保します⇒ほしい人は言ってくださいね。渡し方を考えます。

gurezo, 2016-05-26:
FBメッセンジャーで待ち合わせ場所と時間を決められたらと思いますー。

kohichi000000, 2016-05-26:
> FBメッセンジャーで待ち合わせ場所と時間を決められたらと思いますー。

ですね。そうさせてください。

kazygumi, 2016-05-27:
## 梶井日程
案件次第でまだわからないのですが、可能であればお手伝いしたいです。
- [] 06/08(水) AM
- [] 06/08(水) PM
- [x] 06/09(木) AM
- [x] 06/09(木) PM
- [] 06/10(金) AM
- [] 06/10(金) PM

kazygumi, 2016-05-27:
ここを見て気づいたのですが、A1パネル作りますか？

kohichi000000, 2016-05-27:
@kazygumi さん、できれば！
ちなみに壁が使えることになったので、必ずしもA1でなくてもだいじょぶです。
今後も使えるやつと、今回のみのやつを作るつもりでした。
前者はコミュニティのイメージ的なもので、写真メイン。
後者はInteropは背広系の人が多いので、昨今のIoTと絡めたかたいやつ。
今日素案だけ作ります。

kazygumi, 2016-05-27:
@kohichi000000 さん
わかりました！
もしよろしければ自分の方でデザインしちゃいます。

kohichi000000, 2016-05-27:
@kazygumi さん。あざます。できればそうしていただけると・・・。
A3なら来週中にあれば間に合います。
ちなみに、イメージセンスのない私は、前者（いつでも使えるやつ）については
[image]
の文言を変えたものを作ろうとしていました。
（と思いつつ、Webっぽさが分かる写真も欲しいかも）

文はこんな感じに変更を想定：
CHIRIMEN Open Hardware
CHIRIMEN Open Hardware is an open community developing 
Web of Things environment designed based on the concept:
"Web Developers can develop NOT ONLY Web pages BUT 
Web devices without changing their ordinary developing style".
（サイトとかFBのURLはチラシの時と同じように変更）

kohichi000000, 2016-05-27:
パネル案作りました：
[InteroppanelCHIRIMEN.pptx](https://github.com/chirimen-oh/any-issues/files/286539/InteroppanelCHIRIMEN.pptx)

kazygumi, 2016-05-27:
了解しました！

kohichi000000, 2016-05-29:
あ、先日あげた下記は「背広系の人向け」のやつです：
[InteroppanelCHIRIMEN.pptx](https://github.com/chirimen-oh/any-issues/files/286539/InteroppanelCHIRIMEN.pptx)

kazygumi, 2016-06-02:
遅くなってしまい申し訳ないです。
背広の人向けのデザインを作りました。

[CHIRIMEN-A3パネル-背広の人向け.pdf](https://github.com/chirimen-oh/any-issues/files/296202/CHIRIMEN-A3.-.pdf)

@kohichi000000 さんの
意図と違うものになっていないか、ご確認お願い致します。

kohichi000000, 2016-06-03:
@kazygumi さん
感動です。ありがとうございます。

２点だけ。

○２つ目の箱にある
「IoT アプリを機能分解⇒個別のコンピューティング PF に」
は、「WoTのコンセプト」と「我々の活動」をつなぐ言葉でしたので、
そのような感じになるように書いていただけないでしょうか？

○右下の電灯のところの英語を
Even a hardware HMI will be flexible
としてください。

すいませんが、よろしくお願いいたします。

kazygumi, 2016-06-03:
了解しました！

kazygumi, 2016-06-03:
修正してみましたのでご確認お願い致します

[CHIRIMEN-A3パネル-背広の人向け.pdf](https://github.com/chirimen-oh/any-issues/files/297844/CHIRIMEN-A3.-.pdf)

kohichi000000, 2016-06-03:
ありがとうございます。確認しました。わたしとしてはこれで大丈夫です！
ほんとにありがとうございます！

kazygumi, 2016-06-04:
すみません！
CHIRIMENのイラストが崩れていたので差し替えたいです。

[CHIRIMEN-A3パネル-背広の人向け.pdf](https://github.com/chirimen-oh/any-issues/files/298603/CHIRIMEN-A3.-.pdf)

印刷の際はこちらでお願い致します。

kohichi000000, 2016-06-04:
@kazygumi さん
了解しました。ありがとうございます。

kohichi000000, 2016-06-04:
まとめます。

* A1パネルの印刷は無事SFCさんにお願いしました。私が７日に回収して８日に掲示できる予定
* 展示スタッフ
 - ８日：　@gurezo さん, @kohichi000000
 - ９日：　@kazygumi さん, @kohichi000000
 - 10日：　@kohichi000000
-  後の方は明日確認させていただく予定。
* 展示物は私が遅くとも８日朝に搬入予定
* パンフとシールは７日に開場に届く（はず？）

gurezo, 2016-06-04:
@kohichi000000 さん、すみません
今の状況だと色々怪しいので、自分を頭数から外して下さい。
すみません。

kohichi000000, 2016-06-04:
@gurezo さん
はい、都合がつけばで構いません。

kazygumi, 2016-06-04:
> A1パネルの印刷は無事SFCさんにお願いしました。私が７日に回収して８日に掲示できる予定

あ！A1パネルだったんですね。
A3サイズで作っちゃいましたけど大丈夫でしょうか…。
もし直す必要があればご連絡ください。

> パンフとシールは７日に開場に届く（はず？）

パンフはすでに出荷され、6/6に弊社に届く予定ですので、
6/7の指定の時間に届くように弊社から再発送します。

シールは今のところ問題なく進んでいるようで、
出荷したら出荷情報を教えてくれるそうです。
校正紙が届いていたので、念のため共有します。

[校正紙_W10241_株式会社ブリリアントサービス梶井祐介様.pdf](https://github.com/chirimen-oh/any-issues/files/298964/_W10241_.pdf)

kohichi000000, 2016-06-05:
@kazygumi さん
パネル問題ないですよ。ありがとうございます。

sizuhiko, 2016-06-05:
木PMと金フルで参加しますので、基調講演聞いてない時間とかお手伝いできると思います

kohichi000000, 2016-06-05:
@sizuhiko さんありがとうございます

kazygumi, 2016-06-06:
パンフ、本日届きましたので、幕張メッセに送りました。
また、シールも配送したと連絡がありました。
それぞれの配送情報は下記の通りです。

配送業者：クロネコヤマト（両方とも）
伝票番号：
　チラシ：332756124632
　シール：404236196066

kohichi000000, 2016-06-07:
いいわすれましたが、チラシもステッカーも無事届きました。ありがとうございます！

kazygumi, 2016-06-08:
@kohichi000000 さん
すみません！
仕事でトラブルがおきてしまい明日は行けなさそうです…。
ぜひ行きたかったのですが…。
直前で申し訳ありません。

kohichi000000, 2016-06-08:
@kazygumi さん
だいじょぶです。気になさらないでください。
ほんとは直接お礼したかったですー

kazygumi, 2016-06-10:
皆様
展示お疲れ様でした。
参加することができませんでしたが、
facebookの写真で楽しませて頂きました。

@kohichi000000 さん
こちらこそご挨拶したかったのですが…。無念でした。

今回製作したもろもろのAIファイルを
下記においておきます。
[chirimen-pr160610.zip](https://github.com/chirimen-oh/any-issues/files/308731/chirimen-pr160610.zip)
自由にお使いください。

kohichi000000, 2016-06-12:
皆様のおかげで無事終わりました。
本issueはクローズさせていただきますー。
余談ですが、
* パンフ、ステッカーは私の手元にあります。⇒モジラジャパンさんにお持ちします。
* A1のパネルは慶應においておきます(Maker Faireとかでも使えるかも)

closed 2016-06-12 by kohichi000000