chirimen-oh/any-issues #42: チラシの刷新

kohichi000000, 2016-05-16:
昨年コミュニティで使っていた
[チラシ.pdf](https://github.com/chirimen-oh/any-issues/files/265747/default.pdf)
の刷新案です。5/23あたりにはフィックスしたいですー

# 表面（白い方）

## 左にあるCHIRIMENの写真
- ピンが片側18個のものに変えたほうがよいのでは

## 右にあるプログラム
- promise化前のものなので、promise化されたものに変更すべき？
- 例えば、https://github.com/chirimen-oh/examples/blob/master/LEDblink/js/main.js の中身をそのまま書き出すのはどうか？

## コミュニティ名
- 「CHIRIMEN Open Source Hardware」に変更

# 裏面（黒い方）

## 一番上にあるCHIRIMENの絵
- 絵はAs isで問題ないと思うが、プリントパターンに書かれている文字列について、https://github.com/chirimen-oh/release/blob/master/hardware/CMN2015-1_printPattern.pdf
に書かれている通りに「Open Source Hardware for Web Makers」とすべき？

## 上部の説明：こんな感じはどうか？
- Make Real Things with Browser Technologies
 - CHIRIMEN Board is designed for Web developers to MAKE real things that are connected to the Web via Browser technologies.

- Easy Way to Make
 - GPIO/I2C WebAPIs (to control devices) are introduced into Gecko engine on CHIRIMEN. Web developers can keep their own ordinary development style.

- Open and Neutral
 - Developers can get the firmware code, board design, manufacturing process, API spec, and all necessary sources from the following URL. This community honors developer's will and work on the standeardization.

## Spec
- As isで問題なし。

## URL
- https://chirimen.org

kohichi000000, 2016-05-19:
フォントはアレですがざっくりこんなイメージを想定しています
[チラシ表.pdf](https://github.com/chirimen-oh/any-issues/files/271879/default.pdf)
[チラシ裏.pdf](https://github.com/chirimen-oh/any-issues/files/271880/default.pdf)

gurezo, 2016-05-19:
#### 気になった点　チラシ裏.pdf
##### Specs以下のアンダーラインが中途半端
- CPU
- GPU
- Memory
- Storage
- Interface

kohichi000000, 2016-05-19:
文字に合わせてアンダーラインしておけばOKですかね？

gurezo, 2016-05-19:
はい、文字長に合わせて、アンダーラインが引いてあれば良いかと思います。

naobsd, 2016-05-19:
裏面、NANDは8GBです。
microUSBは一つと書くべきです。あるいはOTGと書くべきです。

kohichi000000, 2016-05-20:
ありがとうございます。確かにそうですね。Specを次のように直します：

- CPU:       RK3066 (ARM Cortex A9 1.6GHz dual core)
- GPU:       Mali 400
- Storage:   DDR3 1GB
- Memory:    NAND Flash 8GB, microSD slot
- Interface: microHDMI, microUSB x1(OTG), USB, GPIO, I2C, etc.

kohichi000000, 2016-05-20:
なんとなくこっちの方が気持ちいいですかね。一気通貫でWebだけでできるというのは大きな特長の１つですし。

Web developers need NOT change their own ordinary development style.
Web開発者は従来の開発スタイルを変える必要がない（新たに言語を勉強する必要がない。同じ感覚でリアルなデバイスも触れちゃう）

kohichi000000, 2016-05-23:
とりあえず、現状版をUPします：
[チラシ裏２.pdf](https://github.com/chirimen-oh/any-issues/files/277052/default.pdf)
[チラシ表２.pdf](https://github.com/chirimen-oh/any-issues/files/277053/default.pdf)

kohichi000000, 2016-05-24:
こちらについて、（フォントを除き）これで24(火)中にフィックスする予定です。
[チラシ表２.pdf](https://github.com/chirimen-oh/any-issues/files/277053/default.pdf)
[チラシ裏２.pdf](https://github.com/chirimen-oh/any-issues/files/277206/default.pdf)
問題あれば突っ込みを。

naobsd, 2016-05-24:
http://www.arm.com/cortex-a9.php 眺めてたんですが、裏のCPUの()の中の表記、 "ARM® Cortex®-A9" とちゃんとした方がいいんですかね...

あと細かいツッコミで申し訳ないですが、 "microUSB x1(OTG)" は "microUSB (OTG)" でよいかなと

naobsd, 2016-05-24:
私のデザインセンスは大変残念なので的外れかもですが、表の方、Cのロゴをもうちょっと主張させて、 "HIRIMEN Hello, Real World" の文字列は少し控えめにするとどうかなと思いました

kohichi000000, 2016-05-24:
ありがとうございます。修正します。
よく見たらMaliもTMなんですね・・・
https://www.arm.com/ja/products/multimedia/mali-gpu/ultra-low-power/mali-400.php
入れておきます。

gurezo, 2016-05-24:
[チラシ表２.pdf](https://github.com/chirimen-oh/any-issues/files/277053/default.pdf)
１８ピンに変えたい
[チラシ裏２.pdf](https://github.com/chirimen-oh/any-issues/files/277206/default.pdf)
ここの文字列を「Open Source Hardware for Web Maker」に変えたい

上記メモを外しておいた方が良い段階かと。
入稿時にメモがあるのは、怖いっす。

kohichi000000, 2016-05-24:
ありがとうございます。これに従い入稿前に絵を差し替えますので。

kohichi000000, 2016-05-24:
表のCのロゴと他の文字のバランスですが、こんな感じでどうでしょ？
[チラシ表2-c1.pdf](https://github.com/chirimen-oh/any-issues/files/279024/2-c1.pdf)

gurezo, 2016-05-24:
画像の下地とチラシの下地の色と合わせれば、
個人的意見ですが、これで良いかと

kohichi000000, 2016-05-24:
はいー下地は無色にしてはりますね

kohichi000000, 2016-05-24:
とりあえず、Lチカの絵はこれに差し替える：
[image]

kohichi000000, 2016-05-24:
とりあえず、絵も差し替えました。こんなんでどうかな（背景消しはまだ）：
[チラシ表3.pdf](https://github.com/chirimen-oh/any-issues/files/279192/3.pdf)
[チラシ裏3.pdf](https://github.com/chirimen-oh/any-issues/files/279193/3.pdf)

kazygumi, 2016-05-24:
あ、すみません！
デザイン的なところまで対応いただいていたのですね。
勝手に文言FIXしたと思い込み、今日の進行を見ずにデザインをしてしまいました。

[CHIRIMEN-A4チラシ.pdf](https://github.com/chirimen-oh/any-issues/files/279803/CHIRIMEN-A4.pdf)

自分からの提案は下記のとおりです。

### 表面
・女子高生の話も出ていたので、水色やイラストを引いて、少しかわいくしました。
・「CHIRIMEN」の文字を小さくして、キャッチコピーから切り離し、商品名っぽくしました。
・コードは、HTMLタグがないとJavaScriptであることがひと目では分かりづらいため、
　HTMLタグを加えました。
・遠目からでもwebエンジニアが反応できるようHTML5のロゴを加えました。

### 裏面
・文字が多かったので、絵的なものを入れて、読みやすくしました。
・チラシの目的・ゴールとしては、手にとった人が
　・サイトをお気に入りに入れてもらう。
　・facebookのグループに参加してもらう。
　・Twitterでフォローしてもらう。
　というところまでアクションしてくれるといいなと思うので
　「CONTACT」としてそれぞれのURLを加えました。

もしよければ部分的にでも採用してもらえると嬉しいです。

kohichi000000, 2016-05-25:
おーすばらしい！さすがというか。基本 kazygumi さんのご提案に賛同します。
（やはり、こういうものはセンスがある人が作ったほうがいいｗ）
いくつかお願いと確認があります。

### 全体
* （確認）サイズですが、A5とかだと小さすぎますかね？（A4だと折られてしまうので昨年度と同じ大きさのA5くらいがうれしい。）

### 表面
* （お願い）右側のプログラムですが、JavaScriptの箇所を下記のようにできないか？（肥大化しちゃいますかね？）
[チラシ表3.pdf](https://github.com/chirimen-oh/any-issues/files/279192/3.pdf)
* （確認）CHIRIMENの絵ですが、写真を使うのと絵を使うのではどっちがいいでしょうかね？写真の方がリアリティが増す、絵の方がかわいらしいとともにメリットはあると思っています。

### 裏面
* （お願い）Easy Way to Makeの2文目を"Web developers need NOT change their own ordinary development style."としてもらえないか？
* （お願い）Open and Neutralの１文目は途中で改行しないほうがいいかも。

余談ですが、裏面のCHIRIMENの特徴を示すロゴは感動しました。すごくわかりやすい。

gurezo, 2016-05-25:
ちょっと見てない間に更に素晴らしいデザインが出てました。
@kazygumi さん、あざます！

kazygumi, 2016-05-25:
賛同いただきありがとうございます！
また、ご指摘ありがとうございます。今日の夜対応しようと思います。

1点確認なのですが、
> （お願い）右側のプログラムですが、JavaScriptの箇所を下記のようにできないか？（肥大化しちゃいますかね？） チラシ表3.pdf
JavaScriptは[チラシ表3.pdf](https://github.com/chirimen-oh/any-issues/files/279192/3.pdf)のコードから、デバック用の処理（console.log、エラー処理）を省いたものなのですが、やはりデバック用の処理も入れておいた方が良いという意図で合っておりますでしょうか。

kohichi000000, 2016-05-25:
たしかに、エラー処理まではいらないですね。
そこまでよみとってくださり、ありがとうございます。

kohichi000000, 2016-05-25:
> （お願い）Open and Neutralの１文目は途中で改行しないほうがいいかも。

わかりにくかったかもしれませんが、こんな感じを意図しています：
Developers can get the firmware code, board design, manufacturing
process, API spec, and all necessary sources from the following URL. 
This community honors developer's will and work on the standeardization.

kazygumi, 2016-05-25:
修正してみました。

■イラスト版
[CHIRIMEN-A5チラシ-イラスト.pdf](https://github.com/chirimen-oh/any-issues/files/281510/CHIRIMEN-A5.-.pdf)

■写真版
[CHIRIMEN-A5チラシ-写真.pdf](https://github.com/chirimen-oh/any-issues/files/281511/CHIRIMEN-A5.-.pdf)

イラスト版の方が気に入っていますが、どうでしょうか。

naobsd, 2016-05-25:
裏のspecのところが戻っちゃってますです

イラストの方が馴染んでいるような...実物も付近に置いてあるような状況で配る用であれば、写真である必要性はあまりないのではないでしょうか(まぁ付近になくても、続きはwebで、見てもらうでいいような)

写真の場合、コピー機で撮ればのっぺりした感じになるんでしたっけ?

kohichi000000, 2016-05-25:
この両者だとイラストの方がいいかも。
写真でトリミングされているとちょっと気持ち悪いのかなぁ。
どうでしょ？

kazygumi, 2016-05-25:
ご指摘ありがとうございます！
SPEC修正しました。

■イラスト版
[CHIRIMEN-A5チラシ-イラスト.pdf](https://github.com/chirimen-oh/any-issues/files/281575/CHIRIMEN-A5.-.pdf)

■写真版
[CHIRIMEN-A5チラシ-写真.pdf](https://github.com/chirimen-oh/any-issues/files/281576/CHIRIMEN-A5.-.pdf)

たしかに、入り口はかわいく見えるイラストで誘導して、
webでリアルな方を見てもらう方が
初心者は入りやすいかもと思いました。

naokisekiguchi, 2016-05-26:
ここで話が進んでいたんですね。。見逃してました。

今更かもしれないですが、サンプルコードの部分、以下のように書けますね。

``` 
'use strict';
window.addEventListener('load', function (){
// WebGPIO LED Blink
  navigator.requestGPIOAccess()
    .then(gpioAccess=>{
      var port = gpioAccess.ports.get(198);
      var v = 0;
      return port.export("out").then(()=>{
        setInterval(function(){
          v = v ? 0 : 1;
          port.write(v);
        },1000);
      });
  });
}, false);  
```

kazygumi, 2016-05-26:
サンプルコードの提供ありがとうございます！
夜に対応します。

kazygumi, 2016-05-26:
サンプルコードを修正しました。

[CHIRIMEN-A5チラシ.pdf](https://github.com/chirimen-oh/any-issues/files/283835/CHIRIMEN-A5.pdf)

こちらで入稿しようと思います。

gurezo, 2016-05-26:
@kazygumi さん、有難うございます！！！

kohichi000000, 2016-05-26:
ありがとうございます。ということで、このissueはcloseしますー

closed 2016-05-26 by kohichi000000