chirimen-oh/any-issues #43: Transfer old issues to chirimen-oh repository

naokisekiguchi, 2016-06-20:
mozopenhardレポジトリのissue
https://github.com/MozOpenHard/CHIRIMEN/issues
で必要なやつはこちらに移してくる必要がある。

何か良い方法ありますか？
なければ地道に移してくる必要がありますが。

gurezo, 2016-06-20:
ちょっと調べただけなので、未確認情報です
[githubのissueを抜き出す(GHE → gitlabへ移行したい)](http://qiita.com/abeyuya/items/7301d01a2906457bdb77)

参考になれば幸いです。

MSakamaki, 2016-06-20:
こういうのもあるぽい(使った事無い)
https://github-issue-mover.appspot.com/

MSakamaki, 2016-07-04:
https://github.com/MozOpenHard の方の権限が無いのでこちらでツール系が使えなさそう。

どなたか @MSakamaki を https://github.com/MozOpenHard のメンバーに追加してもらうとかできますか？
無理そうなら他のツールあたりでexport import試してみます。

dadaa, 2016-07-04:
@MSakamaki 追加いたしましたー

MSakamaki, 2016-07-04:
@dadaa ありがとうございます

１個移してみた。
元をCloseして宛先に自分の名前で新しくopen issuesを立てる感じですね。
(多分どんなツールでも発案者のなりすましは出来ないと思う。)

https://github.com/MozOpenHard/CHIRIMEN/issues/62
to
https://github.com/chirimen-oh/any-issues/issues/53

この方式でよければ、明日Openのモノを全部移しちゃいます
Close含めて引っ越したい場合はもう少し考える必要が有りそうです。

MSakamaki, 2016-07-06:
issues引っ越し残り３つ、
moving issues ラベルをつけました。
(後で名前や色は変えてオッケーです。)

MSakamaki, 2016-07-06:
引っ越し前だけのisseusを表示したい場合はfilterに`is:open is:issue -label:"moving issues" `とか入れる。

MSakamaki, 2016-07-08:
残り3つのissues引っ越しできないと思ったら
ここのメンバーに @YusukeYamamoto  さんと @dynamis さんが居なくて担当アサインできない！って怒られてる事に気がついた。

どうしよ・・・・

gurezo, 2016-07-08:
@YusukeYamamoto さんと @dynamis さん招待sちゃいました（テヘペロ

MSakamaki, 2016-07-08:
引っ越し完了

closed 2016-07-08 by MSakamaki