chirimen-oh/any-issues #44: CHIRIMENのパネル

satakagi, 2016-06-20:
Interop Tokyo用に製作したパネルを更新したいです。
関連するイシューは　#41 
以下が上記パネルの課題と考えられる点です。
・他のプロジェクトにないCHIRIMENの特徴点を明確にしたい（今のコピーだとTesselとの違いがわからない）
・英語版　もしくは日英版　（今のパネルは英語の説明が不完全）

satakagi, 2016-06-20:
変更案を作ってみました。
ただし、変更が必要ない文字の部分は省略してます。（aiファイルの読み込みが失敗してるため・・）

@kazygumi さん、@kohichi000000さん、aiファイルをちゃんと編集できるツールがない（Inkscapeしか持ってない・・・・）ので、原案に対する更新をお願いできないでしょうか？

[CHIRIMEN-A3ポスター改造.pdf](https://github.com/chirimen-oh/any-issues/files/322769/CHIRIMEN-A3.pdf)
）

[CHIRIMEN-A3ポスター改造.zip](https://github.com/chirimen-oh/any-issues/files/322774/CHIRIMEN-A3.zip)

kohichi000000, 2016-06-20:
今日の夜トライしてみますね

kazygumi, 2016-06-20:
@kohichi000000 さん
こちらで対応しましょうか？
今日中は難しいかも知れませんが…

kohichi000000, 2016-06-20:
@kazygumi さん
私の方で無理ならお願いするかも。まずは試してみますね。

kohichi000000, 2016-06-20:
いくつかフォントが無くてダメでした・・・・
@satakagi さんが作られたものに対し、このようにできますか？

○一番上
「センサーやアクチュエータなどの・・・」の下
The embedded computer with a built-in web browser
(computerは単数形でいい？）

○IoTの課題とWoT、および我々の活動の位置づけ
　→IoTの課題とWoT、およびCHIRIMENの位置づけ
　　The position of "Web of Things" and "CHIRIMEN" against ...
（下のキャプションとの関係から、「我々の活動」を「CHIRIMEN」に代える）

○「IoTの課題」の下の文章
こんなんでいかがでしょう？
Although the things network is unified, the applications are still divided.

○「WoTの概念」の下の文章
An IoT Application cooperates ....

○それぞれの絵の中の「プラットフォーム」という日本語ををplatformかP/Fに代えられますか？

○電灯の下のキャプション
・Conventional HMI
・Easy to put any chars
・Easy to localize

kazygumi, 2016-06-20:
@kohichi000000 さん
了解しました！

satakagi, 2016-06-22:
@kazygumi さん、　@kohichi000000 さん、　パネルのほう　いつ頃アップロードいただけますでしょう？　できれば明日いただきたいところです。
　お忙しいところ、せかしてしまうようで申し訳ありませんが。

kohichi000000, 2016-06-22:
申し訳ないですが、あとは @kazygumi さんにお願いするしかないです。
私の手下ではこれ以上は厳しいです・・・。

kazygumi, 2016-06-22:
@satakagi さん
了解しました。今日中にアップロードできると思います。

kazygumi, 2016-06-22:
@satakagi さん、@kohichi000000 さん
取り急ぎアップしましたので確認をお願い致します。
[CHIRIMEN-A3パネル-160622.pdf](https://github.com/chirimen-oh/any-issues/files/327957/CHIRIMEN-A3.-160622.pdf)
[CHIRIMEN-A3パネル-160622.ai.zip](https://github.com/chirimen-oh/any-issues/files/327958/CHIRIMEN-A3.-160622.ai.zip)

satakagi, 2016-06-22:
お忙しいところ、どうもありがとうございました。
内容も今確認した限りばっちりです！！！

ひとまずクローズさせていただきます。

closed 2016-06-22 by satakagi