chirimen-oh/any-issues #45: 回路図についての要望

tadfmac, 2016-06-24:
https://github.com/chirimen-oh/shields/blob/master/CHIRIMENBasicShield/CHIRIMENBasicShield.sch
について、下記改善を検討いただきたく、お願いいたします。

- [ ] 配線クロスしている部分にJunctionを配置してほしいです。端子に複数つなぐのではなく、ラインの途中でJunctionを置いて分岐が視覚的に分かるようにしないと、将来画像でも回路図を公開する場合に、交差しているだけなのか結線しているのか見分けられないので

[image]

- [ ] 回路図にもクレジットを入れませんか？誰が作ったものか分かるように

naokisekiguchi, 2016-07-11:
@tadfmac ありがとうございます。
対応しました!
https://github.com/chirimen-oh/shields/commit/a9741adab65b474345cc0fd8bf54c8bdcffc6a6c
問題なければcloseお願いします。

naokisekiguchi, 2016-07-11:
あークレジットの件はどうしましょう？

tadfmac, 2016-11-04:
> 亀レスすいません。。。。

クレジットの件は特に明記がなければ主張しないということで問題ない気もしますので、一旦Closeします。

closed 2016-11-04 by tadfmac