chirimen-oh/any-issues #51: Maker Faire Tokyo 2016 スタッフ募集

gurezo, 2016-07-03:
## 展示員スケジュールフォーマット
### お手伝い可能な方は、下記フォーマットに追記
- [ ] 08/06(金) AM
- [ ] 08/06(金) PM
- [ ] 08/07(土) AM
- [ ] 08/07(土) PM

MSakamaki, 2016-07-06:
@msakamaki

- [x] 08/06(土) AM
- [x] 08/06(土) PM
- [x] 08/07(日) AM
- [x] 08/07(日) PM

sizuhiko, 2016-07-08:
- [ ] 08/06(土) AM
- [x] 08/06(土) PM
- [x] 08/07(日) AM
- [x] 08/07(日) PM

gurezo, 2016-07-09:
@gurezo 体調に問題なければ、下記日程お手伝いします。
- [x] 08/06(土) AM
- [x] 08/06(土) PM
- [x] 08/07(日) AM
- [x] 08/07(日) PM

kazygumi, 2016-07-26:
今度こそはお手伝い。
- [ ] 08/06(土) AM
- [ ] 08/06(土) PM
- [ ] 08/07(日) AM
- [ ] 08/07(日) PM

noga335, 2016-07-27:
日曜は調整中です。

- [x] 08/06(土) AM
- [x] 08/06(土) PM
- [ ] 08/07(日) AM
- [ ] 08/07(日) PM

naokisekiguchi, 2016-08-01:
なんか全体的に見ることになりそう
- [x] 08/06(土) AM
- [x] 08/06(土) PM
- [x] 08/07(日) AM
- [x] 08/07(日) PM

gurezo, 2016-08-03:
okdさんの代打入力です。
- [x] 08/06(土) AM
- [x] 08/06(土) PM
- [x] 08/07(日) AM
- [x] 08/07(日) PM

gurezo, 2016-08-07:
イベントが終了したので、close

gurezo, 2016-08-07:
イベントが終了したので、close

closed 2016-08-07 by gurezo