chirimen-oh/any-issues #56: ドキュメントの用意

MSakamaki, 2016-07-05:
_From @naokisekiguchi on April 20, 2015 13:20_

MDN上にCHIRIMENに関するドキュメントを用意する。

_Copied from original issue: MozOpenHard/CHIRIMEN#7_

MSakamaki, 2016-07-05:
_From @dynamis on April 20, 2015 13:28_

スマホは Phone Guide ディレクトリの配下に各端末がある:
https://developer.mozilla.org/ja/Firefox_OS/Phone_guide

Board Guide ページを作ってその下に CHIRIMEN ページを作成、Phone Guide から Board Guide へのリンクをするとかいかがでしょう？

https://developer.mozilla.org/ja/Firefox_OS/Board_guide
https://developer.mozilla.org/ja/Firefox_OS/Board_guide/CHIRIMEN

それぞれ親のページで「サブページの作成」をすれば作れるはずです (てか上のような、まだ存在しないページの URL 開いたらもうページ作成画面になる)

MSakamaki, 2016-07-05:
_From @YusukeYamamoto on April 22, 2015 23:33_

取りあえず
https://developer.mozilla.org/ja/Firefox_OS/Board_guide
https://developer.mozilla.org/en-US/Firefox_OS/Board_guide
は作成。
先にjaを作成したところ、en-USへの翻訳リンクが無く翻訳できない状態に・・・
しかたがないので、en-USのURLで新規ページの作成をし、翻訳を行ったが、
今度は
Document with this Slug and Locale already exists.
と表示され、保存できない事態に・・・

なので、
https://developer.mozilla.org/ja/Firefox_OS/Board_guide
https://developer.mozilla.org/en-US/Firefox_OS/Board_guide
の両者は存在するが、相互に翻訳の為のリンク?ができていない。

一旦 ja を削除してから en-US を翻訳しようとしたが、そもそもでページの削除も出来ず。。。

@dynamis さん、 @dadaa さん たすけてください・・

MSakamaki, 2016-07-05:
_From @dynamis on April 23, 2015 2:46_

おおぅ。ja への tranlsation 追加はできなくなっているのですね。
対訳関係のリンク追加をお願いしてみましょう。最悪 ja を削除して翻訳として再作成するばあいでもそれは結局 Admin リクエストが必要ですし。

取り急ぎ IRC で問い合わせ中。まだ返答無し。

MSakamaki, 2016-07-05:
_From @dynamis on April 24, 2015 4:34_

bug: https://bugzilla.mozilla.org/show_bug.cgi?id=1158001

メールでも問い合わせはしておいた。取り敢えず翻訳リンク関係無しのままでも気にせず追加して行って頂ければ良いかと。

MSakamaki, 2016-07-05:
_From @naobsd on April 27, 2015 3:17_

直っているようですね

MSakamaki, 2016-07-05:
_From @dynamis on April 27, 2015 5:5_

https://bugzilla.mozilla.org/show_bug.cgi?id=1158001 で書き込まれているとおり、ja ページの編集画面から「ページのタイトルとプロパティの編集」をクリックして「Find the translation source with the lookup below and then click "Save Changes."」というフィールドで翻訳元 en-US としたいページを選択したらリンク関係は作れるとのことです。

この場合でも ja がオリジナルではなく en-US がオリジナルという扱いになるとい課題はありますが、取り敢えずユーザ目線で言語間リンクは問題なく出来そうですので (MDN が ja オリジナルとかに対応できる夢の未来が到来しない限り) 当面この方法でやっていきましょう。

MSakamaki, 2016-07-05:
_From @YusukeYamamoto on April 27, 2015 11:41_

> ja ページの編集画面から「ページのタイトルとプロパティの編集」をクリックして「Find the translation source with the lookup below and then click "Save Changes."」というフィールドで翻訳元 en-US としたいページを選択したらリンク関係は作れるとのことです。

なるほど、確かに翻訳元のページの指定ができるんですね・・・
※日本語ページが独立してあった際に選択できたか覚えてませんが・・・

とはいえ、次から作成するときは en-US で作成して、そのあと翻訳の方が良いですね。
CHIRIMEN ページの作成しますね

MSakamaki, 2016-07-05:
_From @YusukeYamamoto on April 27, 2015 11:55_

CHIRIMENページに書くことメモ

* CHIRIMEN の概要説明
  - CHIRIMEN とは何ぞや
   →いまは手に入らないよーという旨も忘れずに
  - リンクは Github, Chirimen 公式, Google Group, facebookも?
* How to build 
 - バイナリから焼く方法
   - Linux, Windows, Mac(可能なら)
 - ビルド方法
   - Linux, Windows, Mac(可能なら)
 - 他のデバイスに焼いてお試しする方法も?
  - 書いて良いのか知らないけど OWBとかも?

他にあればコメントください

naobsd, 2016-08-09:
https://github.com/chirimen-oh/any-issues/issues/104#issuecomment-236823504 MDNはいずれ消すとの事なので、こちらはクローズします。
(消すのは #9 で、ドキュメントを用意する話は #104 #74 #73 なり新規なりで)

closed 2016-08-09 by naobsd