chirimen-oh/any-issues #58: install/boot from SD card

MSakamaki, 2016-07-05:
_From @naobsd on April 22, 2015 15:11_

CHIRIMEN の仕様上ブートローダはどうしても NAND フラッシュに書き込んでおく必要があるが、カーネル等を SD カードに書き込んで使う事は技術的には可能。 NAND フラッシュの容量を気にしなくて済むほか、ファームウェアの配布・アップデートも SD カードの書き換えだけであれば比較的楽になると思われる。

ブートローダが Rockchip 独自のものの場合、カーネルおよび initramfs (boot.img) は NAND フラッシュに置くしかないが、他のブートローダであればカーネルを含め SD カードに置く事も可能になるはず。

_Copied from original issue: MozOpenHard/CHIRIMEN#10_

MSakamaki, 2016-07-05:
_From @naokisekiguchi on April 24, 2015 11:10_

製造の際にこちらで用意したイメージを書き込んでおけることが前提となりますが、個人的にはCHIRIMENを手に入れてすぐ使えるという意味でNANDフラッシュにFirefoxOSを含めたシステム全体を格納しておくメリットも大きいと感じています。（raspberry piとの差分にもなる。）
macからの書き込みの問題等もあるので、最終的にファームウェアアップデートがどの程度簡単にできるようになるかにもかかってきますが。

naobsd, 2017-05-25:
https://github.com/mrjayy/u-boot-rk3066/commits/wip/rk3066-6 u-boot mainlineを使ってみたい