chirimen-oh/any-issues #59: flashing firmware from Mac OS X

MSakamaki, 2016-07-05:
_From @naobsd on April 22, 2015 15:21_

ブートローダの更新は Rockchip 製の upgrade_tool (Linux x86) や AndroidTool (Windows) で行う必要がある為、 Mac OS X では (VM 内で Linux/Windows を使用しない限り) 更新できない。

しかしブートローダ以外のイメージ (boot.img, system.img 等) については、 Mac OS X 環境でも更新できるはず。

* [rkflashtool](https://github.com/linux-rockchip/rkflashtool)
  * libusb があればビルドして使えるはず (誰かに試してほしい)
* fastboot
  * ブートローダが U-Boot であれば (& バグを直せば) 使えるはず

(#10 SD カードにインストールできるなら問題はない)

_Copied from original issue: MozOpenHard/CHIRIMEN#11_

MSakamaki, 2016-07-05:
_From @YusukeYamamoto on April 27, 2015 11:48_

> rkflashtool

libusb を入れた後、Mac OS X 10.9.2 (marvericks) では make -> バイナリ が出来ました。
焼くのは後ほどやります。
※申し訳ないですが、GW入ってからの予定です。

MSakamaki, 2016-07-05:
_From @naokisekiguchi on May 14, 2015 13:33_

mac上のVM内にubuntu を入れて焼いているのですが、
・CHIRIMENをVM側に接続してupgrade_toolで焼く
・wait for loader device （うろ覚え）のような表示の際にもう一度CHIEIMENをVM側に接続する
という形で焼いています。
イメージを焼く過程で一度CHIRIMENの認識がHOST PC側に移ってしまっているようで。

本質的な解決策でないですが、参考まで。

MSakamaki, 2016-07-05:
_From @naokisekiguchi on May 14, 2015 13:40_

wait for loader startでした

MSakamaki, 2016-07-05:
_From @naobsd on May 14, 2015 14:59_

upgrade_toolでloaderを焼く際にリセットがかかる(デバイスの抜き差しと同じ状態になる)ので、VMの中でupgrade_toolを使う場合、guestに自動的に見せるような設定がなければもう一度guestに見せる作業が必要になると思います。

naobsd, 2016-08-09:
どなたか rkflashtool を Mac でビルドして動作確認できないでしょうか

sizuhiko, 2016-08-10:
とりあえずコンパイルはできた
```
$ brew install libusb

# 省略...

$ make
gcc -O2 -W -Wall -I/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0 rkcrc.c -o rkcrc  -L/usr/local/Cellar/libusb/1.0.20/lib -lusb-1.0
gcc -O2 -W -Wall -I/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0 rkflashtool.c -o rkflashtool  -L/usr/local/Cellar/libusb/1.0.20/lib -lusb-1.0
gcc -O2 -W -Wall -I/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0 rkunpack.c -o rkunpack  -L/usr/local/Cellar/libusb/1.0.20/lib -lusb-1.0
```

sizuhiko, 2016-08-10:
READMEのusageどおりコマンドを実行してみたが、エラーっぽい
```
$ sudo ./rkflashtool p > parm.bin
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
rkflashtool: info: reading parameters at offset 0x00000000
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = Out): device not responding (code = 0xe00002ed)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): device not responding (code = 0xe00002ed)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): device not responding (code = 0xe00002ed)
rkflashtool: info: size:  0x00000000
rkflashtool: fatal: Write error! Disk full?
```

sizuhiko, 2016-08-10:
リブートは動いた
```
$ ./rkflashtool b
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
rkflashtool: info: rebooting device...
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = Out): device not responding (code = 0xe00002ed)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): device not responding (code = 0xe00002ed)
```

sizuhiko, 2016-08-10:
焼き前のボードで確認

```
$ ./rkflashtool v
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = Out): device not responding (code = 0xe00002ed)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): device not responding (code = 0xe00002ed)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): device not responding (code = 0xe00002ed)
rkflashtool: info: chip version: -..-
```

sizuhiko, 2016-08-10:
`rkflashtool n` は実行してみているが、`interface claimed` が出たあと反応がない...
```
$ ./rkflashtool n
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
```

sizuhiko, 2016-08-10:
- 最新版 `v1.0.21: libusb 1.0.21-rc2` をダウンロード
- libusb をコンパイル
- rkflashtool をコンパイル

結果は変わらなかった。とりあえずここまで

sizuhiko, 2016-08-10:
検証環境は以下のとおりです。

- MacBook Air (13-inch, Early 2015)
- Mac OSX Yosemite (10.10.5)
- Xcode 7.1.1

WhiteHawk-taka, 2016-08-10:
rkflashtool p の結果です
```
rkflashtool: info: rkflashtool v5.1
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
rkflashtool: info: reading parameters at offset 0x00000000
rkflashtool: info: rkcrc: 0x4d524150
rkflashtool: info: size:  0x0000025b
FIRMWARE_VER:4.4.2
MACHINE_MODEL:rk30sdk
MACHINE_ID:007
MANUFACTURER:RK30SDK
MAGIC: 0x5041524B
ATAG: 0x60000800
MACHINE: 3066
CHECK_MASK: 0x80
KERNEL_IMG: 0x60408000
#RECOVER_KEY: 1,1,0,20,0
CMDLINE:console=ttyFIQ0 androidboot.console=ttyFIQ0 init=/init initrd=0x62000000,0x00800000 mtdparts=rk29xxnand:0x00002000@0x00002000(uboot),0x00002000@0x00004000(misc),0x00008000@0x00006000(kernel),0x00010000@0x0000e000(boot),0x00010000@0x0001e000(recovery),0x00002000@0x0002e000(backup),0x00040000@0x00030000(cache),0x00002000@0x00070000(kpanic),0x000c0000@0x00072000(system),-@0x00132000(userdata)
```

WhiteHawk-taka, 2016-08-10:
検証環境です
最新のbrewのrkflashtoolでした
Homebrew 0.9.9 (git revision 850db; last commit 2016-08-08)
Homebrew/homebrew-core (git revision c675; last commit 2016-08-08)

- MacBook Air (11-inch, Mid 2013)
- Mac OSX El Capitan 10.11.6
- Xcode 7.3.1
 - gcc : Apple LLVM version 7.3.0 (clang-703.0.31)

一応brewのlistは以下の通りです
```
atk				glib				libcroco			mikutter			readline
autoconf			gmp				libepoxy			mycli				rkflashtool
automake			gnome-doc-utils			libevent			mysql				ruby-build
cabextract			gnu-getopt			libffi				nettle				sane-backends
cairo				gnutls				libgphoto2			node				scrollkeeper
coreutils			gobject-introspection		libicns				openssl				sl
curl				gsettings-desktop-schemas	libpng				p7zip				sqlite
dfu-util			gtk+				librsvg				pango				tmux
docbook				gtk+3				libtasn1			pcre				unrar
ffmpeg				harfbuzz			libtiff				perl-build			webp
fontconfig			hicolor-icon-theme		libtool				pixman				wget
freetype			icu4c				libusb				pkg-config			x264
gd				imagemagick			libusb-compat			plenv				xvid
gdbm				jasper				libvo-aacenc			postgresql			xz
gdk-pixbuf			jpeg				libxml2				python				zenity
gettext				json-glib			little-cms2			rbenv
git				lame				makedepend			rbenv-gemset
```

naobsd, 2016-08-10:
@WhiteHawk-taka libusbもrkflashtoolも自分でビルドしたものではない、という認識でよいでしょうか?

WhiteHawk-taka, 2016-08-11:
@naobsd その認識で合っています。
libusbもrkflashtoolもbrewからのものです。

sizuhiko, 2016-08-13:
私の持っているもう一台のMac(El Capitan)で確認できたので、追加情報を掲載します。

### 環境

- MacBook Pro (Retina, 13-inch, Late 2012)
- Mac OSX El Capitan 10.11.6
- Xcode 7.3.1
    - gcc: Apple LLVM version 7.3.0 (clang-703.0.31)
- libusb: `brew`からインストール
- rkflashtool: `linux-rockchip/rkflashtool`をコンパイル

### 焼いてあるCHIRIMEN

```
$ ./rkflashtool p
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
rkflashtool: info: reading parameters at offset 0x00000000
rkflashtool: info: size:  0x0000025b
FIRMWARE_VER:4.4.2
MACHINE_MODEL:rk30sdk
MACHINE_ID:007
MANUFACTURER:RK30SDK
MAGIC: 0x5041524B
ATAG: 0x60000800
MACHINE: 3066
CHECK_MASK: 0x80
KERNEL_IMG: 0x60408000
#RECOVER_KEY: 1,1,0,20,0
CMDLINE:console=ttyFIQ0 androidboot.console=ttyFIQ0 init=/init initrd=0x62000000,0x00800000 mtdparts=rk29xxnand:0x00002000@0x00002000(uboot),0x00002000@0x00004000(misc),0x00008000@0x00006000(kernel),0x00010000@0x0000e000(boot),0x00010000@0x0001e000(recovery),0x00002000@0x0002e000(backup),0x00040000@0x00030000(cache),0x00002000@0x00070000(kpanic),0x000c0000@0x00072000(system),-@0x00132000(userdata)

$ ./rkflashtool v
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
rkflashtool: info: chip version: 300A-2011.11.11-V101

$ ./rkflashtool n
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
rkflashtool: info: Flash ID: ad de 94 eb 74
rkflashtool: info: Flash Info:
	Manufacturer: Hynix (2)
	Flash Size: 8528MB
	Block Size: 4096KB
	Page Size: 16KB
	ECC Bits: 40
	Access Time: 32
	Flash CS: <0>
```

### まだ焼いていないCHIRIMEN（MFT購入品）

```
$ ./rkflashtool p
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
rkflashtool: info: reading parameters at offset 0x00000000
rkflashtool: info: size:  0x00000286
FIRMWARE_VER:4.4.2
MACHINE_MODEL:marsboard
MACHINE_ID:007
MANUFACTURER:HAOYU-Electronics
MAGIC: 0x5041524B
ATAG: 0x60000800
MACHINE: 3066
CHECK_MASK: 0x80
KERNEL_IMG: 0x60408000
#RECOVER_KEY: 1,1,0,20,0
CMDLINE:console=ttyFIQ0 androidboot.console=ttyFIQ0 init=/init initrd=0x62000000,0x00800000 mtdparts=rk29xxnand:0x00002000@0x00002000(misc),0x00006000@0x00004000(kernel),0x00006000@0x0000a000(boot),0x00010000@0x00010000(recovery),0x00020000@0x00020000(backup),0x00040000@0x00040000(cache),0x003FC000@0x00080000(userdata),0x00002000@0x0047C000(metadata),0x00002000@0x0047E000(kpanic),0x00180000@0x00480000(system),-@0x00600000(user)

$ ./rkflashtool v
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
rkflashtool: info: chip version: 300A-2011.11.11-V101

$ ./rkflashtool n
rkflashtool: info: rkflashtool v5.2
rkflashtool: info: Detected RK3066...
rkflashtool: info: interface claimed
rkflashtool: info: Flash ID: ad de 94 eb 74
rkflashtool: info: Flash Info:
	Manufacturer: Hynix (2)
	Flash Size: 8528MB
	Block Size: 4096KB
	Page Size: 16KB
	ECC Bits: 40
	Access Time: 32
	Flash CS: <0>
```

sizuhiko, 2016-08-13:
Yosemiteではダメですが、El Capitan では大丈夫そうですね。

naobsd, 2016-08-13:
ひとまず El Capitan で libusb を brew からインストールした環境であれば問題ないという事は間違いないようですね。ありがとうございます。

rkflashtool を各自ビルドしてもらうのも面倒かと思うので、最新版のバイナリを https://github.com/chirimen-oh/CHIRIMEN-tools にでも置いて、 brew から libusb をインストールする手順を書いておければいいのかなと思います。 (書く必要ないくらい当たり前の事だったりしますでしょうか?)

----

libusb が悪い可能性も考えてみたのですが、 @sizuhiko さんの Yosemite 環境では、 libusb を brew からインストールしても自前でビルドしても、ダメだったわけですよね。

brew に rkflashtool が登録された (ですよね ?) 時代の OS X では動いていたのではなかろうかと思うので、古い OS X ではダメという事はないと思いたいのですが...

一般的に、 El Capitan で作ったバイナリを Yosemite などの古い OS X で動作させる事は簡単にできるのでしょうか ? できるとしたら、その場合動作するのかどうか、気になります。

naobsd, 2016-08-13:
ところで rkflashtool の使い方というか機能なのですが、書き込めるイメージに制限がある為、何も書かれていない・違うブートローダが書かれているボードは一度 AndroidTool/upgrade_tool (Windows/Linux) で正しいイメージを書き込む必要があります。

一度正しいイメージを書き込んだ状態であれば、ローダーモード (recover ボタン or `adb reboot bootloader`) で起動して `rkflashtool w (partition) < (image)` でイメージを書き込む事ができます。

この時書き込むイメージは、 https://github.com/chirimen-oh/release/releases で配布している pack したイメージではなく、パーティション別のイメージ (boot.img, kernel.img, recovery.img, system.img) です。

パーティション別のイメージは、 B2G をビルドするか、 pack 済みイメージを unpack する事で作れます。

unpack は https://github.com/linux-rockchip/rkflashtool/blob/master/rkunpack.c でできると思うのですが、私が書いた後に修正されているので、ちょっとよくわかりません。私は普段自前の https://github.com/naobsd/rkutils/blob/master/rkunpack.c を使っています。

sizuhiko, 2016-08-13:
> El Capitan で作ったバイナリを Yosemite などの古い OS X で動作させる事は簡単にできるのでしょうか ? できるとしたら、その場合動作するのかどうか、気になります

これは一回私のEl CapitanでコンパイルしたものをYosemiteで動かしてみます。

> パーティション別のイメージは、 B2G をビルドするか、 pack 済みイメージを unpack する事で作れます。

これも後でunpackしてみます。成功して`パーティション別のイメージ`が全て展開されれば成功と思って良いでしょうか？

naobsd, 2016-08-13:
わたしの rkunpack だと

```
$ rkunpack CMN2015-1_B2GOS-20160808.img
VERSION:4.0.2

unpacking
0x00000000 CMN2015-1_B2GOS-20160808.img-HEAD 102 bytes
0x00000066 CMN2015-1_B2GOS-20160808.img-BOOT 194090 bytes
0x0002f690 CMN2015-1_B2GOS-20160808.img-update.img 347199492 bytes

unpacking CMN2015-1_B2GOS-20160808.img-update.img
================================================================================
FIRMWARE_VER:4.4.2
MACHINE_MODEL:rk30sdk
MACHINE_ID:007
MANUFACTURER:RK30SDK

unpacking 9 files
-------------------------------------------------------------------------------
0x00000800 package-file:package-file 639 bytes
0x00001000 bootloader:RK30xxLoader(L)_V2.09.bin 194090 bytes
0x00030800 parameter:parameter:0x2000@0x0 574 bytes
0x00031000 misc:Image/misc.img:0x2000@0x2000 49152 bytes
0x0003d800 kernel:Image/kernel.img:0x8000@0x4000 8384560 bytes
0x0083d000 boot:Image/boot.img:0x10000@0xc000 1107964 bytes
0x0094b800 recovery:Image/recovery.img:0x10000@0x1c000 21528576 bytes
0x01dd4000 system:Image/system.img:0xc0000@0x70000 315920384 bytes
---------- backup:RESERVED:0x2000@0x2c000 0 bytes
-------------------------------------------------------------------------------
================================================================================

0x14b4ce94 CMN2015-1_B2GOS-20160808.img-MD5 32 bytes
unpacked
```

こんな感じで Image/ の下に *.img ができます。

sizuhiko, 2016-08-14:
> El Capitan で作ったバイナリを Yosemite などの古い OS X で動作させる事は簡単にできるのでしょうか ? できるとしたら、その場合動作するのかどうか、気になります

試してみました。バイナリは動きますが、結果はYosemiteでコンパイルしたのと同じで、期待する結果は得られませんでしt。なので、El Capitanが必須環境と言えそうです。

unpackしてみました。
```
$ ./rkunpack CMN2015-1_B2GOS-20160808.img 
rkunpack: info: RKFW signature detected
rkunpack: info: version: 4.4.2
rkunpack: info: date: 2016-08-08 15:33:26
rkunpack: info: family: rk30xx
rkunpack: info: 00000066-0002f68f BOOT                       (size: 194090)
rkunpack: info: 0002f690-14b4ce93 embedded-update.img        (size: 347199492)
unpacked

$ ./rkunpack embedded-update.img 
rkunpack: info: RKAF signature detected
rkunpack: info: file size matches (347199492 bytes)
rkunpack: info: manufacturer: RK30SDK
rkunpack: info: model: rk30sdk
rkunpack: info: number of files: 9
rkunpack: info: 00000800-00000800 package-file               (size: 639)
rkunpack: info: 00001000-0000105e RK30xxLoader(L)_V2.09.bin  (size: 194090)
rkunpack: info: 00030800-00030800 parameter                  (size: 586)
rkunpack: info: 00031000-00031018 Image/misc.img             (size: 49152)
rkunpack: info: 0003d800-0003e7fe Image/kernel.img           (size: 8384560)
rkunpack: info: 0083d000-0083d21c Image/boot.img             (size: 1107964)
rkunpack: info: 0094b800-0094e110 Image/recovery.img         (size: 21528576)
rkunpack: info: 01dd4000-01df9a92 Image/system.img           (size: 315920384)
rkunpack: info: 00000000-ffffffff RESERVED                   (size: 0)
unpacked
```

2度unpackする必要はありましたが、unpackはできたようです。
`parameter`のサイズが違うのは気になりますが...

WhiteHawk-taka, 2016-08-14:
## 手順
- 必須環境
 - Mac OSX ElCapitan
 - 既にB2Gが書き込まれたCHIRIMEN

1. https://github.com/naobsd/rkutils/blob/master/rkunpack.c をgccでビルド
2. リリースイメージをダウンロード + unzip ( https://github.com/chirimen-oh/release/releases )
3. 1のバイナリで2のイメージをunpack
4. brewでrkflashtoolをインストール(同時にlibusbはインストールされる)
5. CHIRIMENをrecoverボタンでブート
6. rkflashtool w system < system.img のように順にイメージを書き込んでいく

基本的にはこんな流れでしょうか？

naobsd, 2016-08-15:
@WhiteHawk-taka さん、ありがとうございます。基本的にはその内容で大丈夫だと思います。

環境について、 El Capitan 以前の ver. も可能であればサポートしたいですが、少なくとも私はどうしようもありませんので、現状としては El Capitan 使ってくださいというしかなさそうですね... ざっくり調べた所そろそろ El Capitan リリースから一年のようですが、 Yosemite もまだまだ現役なのでしょうか?
Windows も 10 以外での動作確認はたぶん (積極的には) しないと思いますので、最新 ver. では動作確認済み、という考え方でよいのかなとは思います。

rkunpack は rkflashtool 同梱のものだと二回必要なようですし、 parameter ファイルにもしかしたらヘッダが付いている気がするので、扱いにくいと思います。 rkutils の方の rkunpack のバイナリを配るようにしたいと思いますので、 https://github.com/chirimen-oh/CHIRIMEN-tools の下に、 "Mac OS X" か何か適当に掘って push していただけないでしょうか?
(面倒でしたら slack の #porting にアップロードしていただければ、こちらでやります)

rkflashtool は、 brew のものは r/w コマンドに partname が指定できるでしょうか? 古いものは offset nsectors しか対応しておらず、使いにくいです。 brew の rkflashtool をどうすれば新しくできるのかわからないのですが、とりあえずこれも最新のバイナリを配りたいと思います。

焼くイメージは、アップデート時は kernel.img boot.img recovery.img system.img の四つ、 /cache /data を初期化したいときは misc.img を(も) 焼く、ですね。焼いた後コマンドで再起動させる場合は "rkflashtool b" です。 (電源抜き差しでも可)

naobsd, 2016-08-15:
@sizuhiko さん、 rkflashtool 同梱の rkunpack で parameter のサイズが違うというのは、コマンドの出力メッセージ (586) と実際のファイルのサイズ (574) が違う、という事でしょうか?

parameter ファイルは PC 上では単なるテキストファイルとして扱う (事が多い) のですが、フラッシュメモリ上に書くデータとしては、 8 バイトのヘッダ(magic+size)と、 4 バイトのCRCが最後に付きますので、 12 バイト差があります。

rkflashtoolだと、p/Pではテキストで扱い、r/wでparameterパーティションを直接読み書きする場合はヘッダ・CRC付きで扱う、となります。
(ちなみにkernel.imgにも同様のヘッダ・CRCが付いていたりします)

WhiteHawk-taka, 2016-08-15:
@naobsd おそらく @sizuhiko さんがやっていたのはrkutils の方のビルドしたrkunpackだと思うのですが、どうなんでしょうか？

sizuhiko, 2016-08-15:
帰宅して実際のファイルサイズを見てみたら
```
-rw-r--r--   1 hoge 501        639  8 14 17:05 package-file
-rw-r--r--   1 hoge 501        574  8 14 17:05 parameter
```
となっていて、parameterは`574`でした。CRCで12バイト差になるのですね。承知しました。

gurezo, 2016-08-23:
###  @gurezo 環境

- MacBook Pro (Retina, 13-inch, Early 2015)
- Mac OSX El Capitan 10.11.6
- Xcode 7.3.1
    - gcc: Apple LLVM version 7.3.0 (clang-703.0.31)
- libusb: `brew`からインストール
- rkflashtool: `brew`からインストール
- rkflashtool: 上記作業後に `linux-rockchip/rkflashtool`をコンパイル

### 焼いてあるCHIRIMEN
### 手順
1. https://github.com/naobsd/rkutils/blob/master/rkunpack.c をgccでビルド
2. リリースイメージをダウンロード + unzip ( https://github.com/chirimen-oh/release/releases )
3. 1のバイナリで2のイメージをunpack
4. brewでrkflashtoolをインストール(同時にlibusbはインストールされる)
5. CHIRIMENをrecoverボタンでブート
6. rkflashtool w system < system.img 
7. rkflashtool w boot < boot.img 
8. rkflashtool w kernel < kernel.img 
9. rkflashtool w misc < misc.img 
10. rkflashtool w recovery < recovery.img 

----

上記作業で、正常に起動しました。

naobsd, 2016-09-07:
何も書かれていないボードはMac OS X上で認識されない、という事について、具体的にどのようにその問題を確認したのか、教えていただけないでしょうか?

また、その時 rkflashtool (githubのソースからビルドしたもの) を実行すると (例えば "rkflashtool v") 、どのような表示がされるのでしょうか?