chirimen-oh/any-issues #61: prepare images for flashing

MSakamaki, 2016-07-05:
_From @naobsd on April 26, 2015 1:2_

after ./build.sh, images for flashing need to be prepared as follows

* out/target/product/chirimen/kernel
  * rkcrc -k kernel path/to/Image/kernel.img
* out/target/product/chirimen/ramdisk.img
  * rkcrc -k ramdisk.img path/to/Image/boot.img
* out/target/product/chirimen/recovery.img
  * cp recovery.img path/to/Image/recovery.img
* out/target/product/chirimen/system.img
  * out/host/linux-x86/bin/simg2img system.img tmp.img
  * resize2fs tmp.img 384M
  * cp tmp.img path/to/Image/system.img

these commands should be run in flash.sh.

_Copied from original issue: MozOpenHard/CHIRIMEN#13_

MSakamaki, 2016-07-05:
_From @naobsd on April 27, 2015 1:55_

commands for flashing:

    $ upgrade_tool di -k kernel.img
    $ upgrade_tool di -b boot.img
    $ upgrade_tool di -r recovery.img
    $ upgrade_tool di -s system.img

MSakamaki, 2016-07-05:
_From @naokisekiguchi on May 18, 2015 10:58_

Preparing 'rkcrc' command. 

http://linux-rockchip.info/mw/index.php?title=Building_Kernel_for_Linux_Distribution
$ git clone --depth 1 https://github.com/naobsd/rkutils
$ cd rkutils
$ gcc rkcrc.c -o rkcrc

Is it OK?

MSakamaki, 2016-07-05:
_From @naobsd on May 19, 2015 0:29_

@naokisekiguchi yes, but

    $ git clone https://github.com/naobsd/rkutils
    $ cd rkutils
    $ make rkcrc

should be more generic.

(btw that wiki contains a lot of errors, please do not rely that so much...)

naobsd, 2016-07-08:
refer #70

closed 2016-07-08 by naobsd