chirimen-oh/any-issues #67: Chirimen Default Starter App

MSakamaki, 2016-07-05:
_From @arky on August 29, 2015 8:58_

It would be a good idea to replace default 'Homescreen' app in the pre-built image with custom Chirimen default app. 

Here are few ideas for the default starter app: 
* Will provide a screen with some information about Chirimen. 
* It will run a basic blink example 
* An input UI element on the page will enable users to change the rate of blink (in seconds)
* Users who connected to Chirimen using the Firefox remote connection manager, could change the blink rate using Devtools console 
* If certain component (a gyro) is connected on pre-assigned GPIO ports then default app will use    Screen.orientation Web API. 

_Copied from original issue: MozOpenHard/CHIRIMEN#26_

MSakamaki, 2016-07-05:
_From @naobsd on August 29, 2015 11:30_

:+1:

MSakamaki, 2016-07-05:
_From @stretchyboy on August 30, 2015 8:47_

:+1:

MSakamaki, 2016-07-05:
_From @autonome on August 30, 2015 17:45_

+1. The smartphone Homescreen is not very useful for IoT development case.
Probably still need a way to access some basics like the Settings app.

MSakamaki, 2016-07-05:
_From @arky on September 9, 2015 7:31_

Most embedded hardware (ie.. OpenWRT, Arduino Yun) exposes the settings via a onboard web server. You just point your browser to particular port to access the settings UI.  

Here is how Yun does it: 
https://www.arduino.cc/en/Guide/ArduinoYun#toc14

MSakamaki, 2016-07-05:
_From @dynamis on November 26, 2015 2:35_

Note about current workaround:

You can insert simple script to launch specific app in homescreen.

Please insert ```<script defer="" src="js/launchapp.js"></script>``` tag in index.html (at end of HEAD or BODY tag) of homescreen
https://github.com/mozilla-b2g/gaia/blob/master/apps/homescreen/index.html
and add apps/homescreen/js/launchapp.js file like this:

https://gist.github.com/dynamis/cb25a9b743a7a79401f4

If you want to launch your original app, please specify origin in your app manifest. Then the app id will be the origin, not random UUID and you can specify the app easily.

naobsd, 2017-02-05:
必要があれば #147 で議論する

closed 2017-02-05 by naobsd