chirimen-oh/any-issues #70: Fails on generate image file on flash.sh

MSakamaki, 2016-07-05:
_From @YusukeYamamoto on November 19, 2015 10:31_

[ problem ] 
In MozOpenHard repos, clone git repo. config it, build it, and then try to generate image to exec flash.sh.
We can't get image file.

[ cause ]
Chirimen's section is missing on flash.sh
We have to add bellows :

```sh
 +"chirimen")
 +  cp device/rockchip/rksdk/buildspec-rk3066.mk ./buildspec.mk
 +  device/rockchip/rksdk/mkimage.sh
 +  rm ./buildspec.mk
 +  rkst/mkkrnlimg out/target/product/chirimen/kernel rockdev/Image/kernel.img
 +  exit 0
 +  ;;
```

_Copied from original issue: MozOpenHard/CHIRIMEN#31_

MSakamaki, 2016-07-05:
_From @naobsd on November 19, 2015 13:11_

see also: #13

MSakamaki, 2016-07-05:
_From @YusukeYamamoto on January 18, 2016 7:24_

対応済み。
ただし、2.5は正式対応ではないので、CLOSEさせないでおく。
※正式対応ではない = 誰が決定権もっているかわからないとも言う。

naobsd, 2016-08-09:
対応済みで、現状既にv2.5でイメージをリリースしている為、クローズ。

closed 2016-08-09 by naobsd