chirimen-oh/any-issues #71: 最新のボードの綺麗な写真を用意する

MSakamaki, 2016-07-05:
_From @dynamis on December 1, 2015 9:5_

ちゃんとしたカメラで撮影したボードの画像が素材として必要

_Copied from original issue: MozOpenHard/CHIRIMEN#32_

MSakamaki, 2016-07-05:
_From @dynamis on December 1, 2015 9:7_

4k TVでじっくり見られても恥ずかしくない解像度でおねがいします ;) by naobsd

MSakamaki, 2016-07-05:
_From @kudodo on December 5, 2015 9:34_

http://web.sfc.wide.ad.jp/~kudo/tmp/CHIRIMEN600dpi.zip

こんなのでいかがですか？
ボードの汚れというか傷？がちょっときになるのでキレイなボードを今度さがします。

MSakamaki, 2016-07-05:
_From @dynamis on January 20, 2016 10:47_

ひとまず十分かと思います
https://github.com/MozOpenHard/CHIRIMEN-tools/tree/master/photoImages
に追加してこの issue は close いただければ

naobsd, 2016-07-08:
mozopenhard的な文字列が入っていない写真が必要？

naobsd, 2016-08-09:
https://chirimen.org/docs/ja/board_connectors.html この裏面の写真を置き換えたい

naobsd, 2017-02-05:
https://chirimen.org/docs/ja/board_connectors.html については #111 で対応されている