chirimen-oh/any-issues #74: Prepare english documents for build and install

MSakamaki, 2016-07-06:
_From @dynamis on February 2, 2016 6:59_

We currently have this page in wiki
https://github.com/MozOpenHard/CHIRIMEN/wiki/Setup

but it don't contain enough/full information like this:
https://github.com/MozOpenHard/CHIRIMEN-tools/blob/master/doc/installFirefoxOS.pdf

let's prepare english documents to work with global contributors.
I believe this document should be prepared before the general availability (-> Milestone 1).

_Copied from original issue: MozOpenHard/CHIRIMEN#58_

MSakamaki, 2016-07-06:
_From @dynamis on February 2, 2016 7:6_

Better to write in MDN?
https://developer.mozilla.org/en-US/Firefox_OS/Board_guide/CHIRIMEN
or write in our Wiki and link from MDN?

I prefer to write in MDN since it support l10n and we can find relating information about porting to other boards (will coming soon?) within the same site.

naobsd, 2017-02-05:
handle this on #162

closed 2017-02-05 by naobsd