chirimen-oh/any-issues #75: SDカードの有効化

MSakamaki, 2016-07-06:
_From @YusukeYamamoto on January 26, 2016 10:5_

CHIRIMENでSDカードを使えるように有効化する。

https://docs.google.com/document/d/15CJ2rqeY_AfzI8l3IMFRO48N7Y1cAg9etx_JyPLzwKc/edit
の「深海メモ」も参照

Issue #16  と類似

_Copied from original issue: MozOpenHard/CHIRIMEN#57_

MSakamaki, 2016-07-06:
_From @ashie on February 14, 2016 7:20_

Firefox OS初心者かつCHIRIMENが何なのかもよく分かっていませんが、状況を調べてみて、一応有効化
できたように見えるのでメモしておきます（このIssueは日本語で書かれているので日本語で続けておきます）。

まず、SDカードを挿入して`navigator.getDeviceStorages("sdcard")`などでアクセスするとSDカードには
一通りアクセス出来た(書き込みもOK)ため、GaiaのSettingsアプリでの判別に問題があると考えました。

GaiaでMedia Storageを有効化/無効化しているのは以下辺りと思われます。

https://github.com/MozOpenHard/gaia/blob/v2.1-mozopenhard/apps/settings/js/panels/root/storage_usb_item.js#L191

デフォルトストレージの設定は`device.storage.writable.name`を見ているようです。

https://github.com/MozOpenHard/gaia/blob/v2.1-mozopenhard/apps/settings/js/panels/root/storage_usb_item.js#L29

実機のprefs.jsではこの値が`internal_sd`となっていますが、これは内部ストレージのuserパーティション
を指しています（SDを指定したい場合はexternal_sd)。

https://github.com/MozOpenHard/device-rockchip-rksdk/blob/mozopenhard/fstab.rk2928board.bootmode.unknown#L11

userパーティションは、現在のパーティション構成では存在しないようです。

https://github.com/MozOpenHard/CHIRIMEN-tools/blob/master/rockdev-rk3066/parameter#L11

このため、Media Storageの設定が無効化されているようです。

私は最近いじり始めたばかりで本プロジェクトの方針を把握していないため、userパーティションを作る
べきなのか、それともデフォルトストレージを変更すべきなのか判断がつかなかったのですが、ひとまず
後者をトライしてみました。

デフォルトストレージの設定は以下のパッチで変更しているようです。

https://github.com/MozOpenHard/b2g-patches/blob/v2.1-kk/0004-gaia-Set_device.storage.writable.name_to_internal_sd.patch#L12

これを`external_sd`に変更したところ、SettingsアプリでもMedia Storageが有効化されました。

現状ではinternal_sdが使えないのにも関わらず`getDeviceStorages("sdcard")`で取れてしまうため、
取れないようにするか、使えるように修正する(内部ストレージを割り振る)などの処置も必要と思われます。

MSakamaki, 2016-07-06:
_From @naobsd on February 14, 2016 7:58_

fstabはfstab.rk30board.bootmode.unknownの方になります。

個人的には、

* 内蔵ストレージもSDカードとして使えるようにする方がよい
 * ただしuserパーティションを復活させるのではなく、/data/mediaをemulatedなSDカードとするやり方の方がよい https://source.android.com/devices/storage/config-example.html#android_5_x_both
* device.storage.writable.nameはRockchipの流儀に合わせるのではなく、b2gデフォルト(sdcard)に合わせる方がよい
 * b2g v2.0の時はsdcard, sdcard1というラベルにすれば両方認識した https://github.com/MozOpenHard/device-rockchip-rk30sdk/commit/6cba821fffaa21f026f2e648ccbd1c8eebfea319
 * device.storage.writable.nameは元に戻してvoldmanaged=を変えればよい? https://github.com/MozOpenHard/device-rockchip-rksdk/blob/mozopenhard/fstab.rk30board.bootmode.unknown#L12

と思っています

MSakamaki, 2016-07-06:
_From @ashie on February 14, 2016 8:19_

> fstabはfstab.rk30board.bootmode.unknownの方になります。

あ、そうですね。失礼しました。

> 内蔵ストレージもSDカードとして使えるようにする方がよい

確かに、この方が内蔵ストレージを効率的に利用できて良いですね。

> device.storage.writable.nameはRockchipの流儀に合わせるのではなく、b2gデフォルト(sdcard)に合わせる方がよい 

私もそう思います。

MSakamaki, 2016-07-06:
_From @ashie on February 14, 2016 11:54_

補足します。

> device.storage.writable.nameは元に戻してvoldmanaged=を変えればよい?

たぶん、これは必要無いと思います。
現状でも、メディアタイプ`sdcard`でinternal_sdもexternal_sdもリストとしては返ってきているからです。

このissueの原因はGaiaのストレージ関連共通コードの以下の挙動だと考えています。

* prefs.jsで特に指定していない場合やデフォルトの`sdcard`の場合、getDevciceStorages()の先頭のストレージをデフォルトストレージとして扱う
    * https://github.com/MozOpenHard/gaia/blob/v2.1-mozopenhard/apps/settings/js/panels/root/storage_usb_item.js#L252
    * CHIRIMENの場合はprefs,jsで指定しているのでinternal_sdとなるが、リストの先頭はinternal_sdなので`sdcard`の場合も結局internal_sdになる
* デフォルトストレージが無効の場合、他に有効なストレージがあってもMedia Storageの設定を無効化してしまう
    * （Media Storage設定が有効化されれば複数のストレージも扱える）

まとめると

* internal_sdさえ有効にすればexternal_sdもMedia Storageの設定で見られるはず
* prefs.jsのパッチはinternal_sdさえ有効なら本来は不要なので削除した方が良い
* external_sdを使うことはアプリ側で直接指定すれば現状でも出来ている
* Gaia標準アプリは上記の共通コードを使ってしまうため現状ではexternal_sdを使えない

MSakamaki, 2016-07-06:
_From @naobsd on February 14, 2016 12:47_

なるほど、ありがとうございます。とりあえずinternal_sdを(どんな形にせよ)使えるようにすればよさそうですね。

naobsd, 2017-02-04:
昔internal_sdだったuserパーティションはuserdataパーティション(/data)に統合してしまっていて、パーティションの切り直しは面倒だと思います。

/dataの一部を仮想SDカードにする方がよいと思います
 https://source.android.com/devices/storage/config-example.html#android_5_x_both

naobsd, 2017-02-05:
#63 も直りますかね?