chirimen-oh/any-issues #85: センサ値などの取得に関する課題(Bugzillaでの指摘を具体例でわかりやすいように記載)

MSakamaki, 2016-07-06:
_From @YusukeYamamoto on January 18, 2016 8:37_

tl;dr

https://bugzilla.mozilla.org/show_bug.cgi?id=1166588
にて指摘頂いている課題の具体例の記載。
♯ 指摘内容と、それがアプリに対してどういう制限として繋がるのかを明確にする。

現状のGPIO APIの実装では、センサ(及びボタンなどのデバイス)からの値を読み取るためには、readをポーリングで処理せねばならない。

そのため、アプリケーション層から、特定デバイスに対して一定周期でのボーリングにて実装する必要がある。
つまり、センサ(やボタンなど)から値を細かい周期で取ろうと思えば思うほど、UIブロックが発生してしまう。

JavaScriptはシングルスレッドなのでこれは仕方がない。
なので、Workerスレッドに逃せば出来るかと想像できるが、課題が存在。

Geckoの実装として、Gaiaのアプリ上からWorkerスレッドを呼べるようにするにはGecko側にWorkerから呼べるように修正が必要。現状の実装ではWorkerから呼べるようにしていないので、GPIO/I2CはWokerから呼べない( (試していないため、確認は必要だが )。
※ ここで書いているのは、Kyleから指摘されているe10s/IPCとは別な話。WebAPIとして、Gaiaのアプリケーション層から操作するための話。

なので、現状の結論からでは、以下の制限が発生。
　例えば、CHIRIMENに加速度センサを付けて傾きに応じて、LEDにを光らす / UIを更新するアプリ。
　傾き検知の値を細かい値にすればするほど、メインスレッドでポーリングが必要。
　そのため、メインスレッドにてブロッキングが発生。
　LEDを光らすのはもちろん、描画更新も出来ないし、何もできなくなる。
　センサ値取得処理をWorkerスレッドに逃そうにもWorkerスレッドからは呼べない(はず)なので、現時点においては、回避方法はない。
　→要はメインスレッド内で処理できるキャパシティが決まっているので、センサをセンシティブに取得して、DOM操作、他のGPIO/I2C操作など凝った動作をしようと思うと制限が出てくる。「センサ取得を荒くして、DOM操作に割く」、「センサ取得がセンシティブだが、DOM操作は制限」など、作りこみが大きくなればなるほど影響が出てくる。

_Copied from original issue: MozOpenHard/CHIRIMEN#47_

MSakamaki, 2016-07-06:
_From @dadaa on January 18, 2016 12:17_

はい，Gaia アプリ上の Worker から呼べないことを確認しました．
実装としては現在 navigator に備わっている mozGpio ないし mozI2c を WorkerNavigator ( https://developer.mozilla.org/ja/docs/Web/API/WorkerNavigator ) に実装するか，Gecko のメソッド内で worker スレッドを作るかでしょうか．

MSakamaki, 2016-07-06:
_From @YusukeYamamoto on January 18, 2016 14:16_

私もworkerにAPI追加したことは無いので、さきほどコードを読んだ範囲でですが、

Gaiaのアプリ上でWorkerから呼べるようにするためには
実APIをWebIDLで定義、WorkerNavigatorのWebIDLからも定義、worker側への実装が入り口(API)として最低限必要みたいです。

その上で、実体をGecko側に実装が必要。これはKyleから指摘もらっているようにe10s/IPCを用いて実装し、非同期で処理できるようにすることが、Workerとしてmustか否かは不明。不要なら今のままでもとりあえず動くはず。
※APIの口として、Gecko/Gaia のつなぎ部分のみworkerとしての実装が必要なのか、はたまた処理内部も分ける必要があるのか。こちらは要確認。できればMozillaさん内部で確認してくれたほうが確実な情報になりますし、確度あがるのでみなさん嬉しいと思いますが。

ですので、ご質問については、
> 実装としては現在 navigator に備わっている mozGpio ないし mozI2c を WorkerNavigator ( https://developer.mozilla.org/ja/docs/Web/API/WorkerNavigator ) に実装するか，

→半分Yesです。それだけだと対応不足。この対応はmsut。これしないとGaiaのアプリから呼べない。

> Gecko のメソッド内で worker スレッドを作るかでしょうか．

→多分No。

あとは、本制限によってどの程度アプリ作成に制約が出てくるかですかねぇ。
ハードに詳しくないのでどの程度の解像度(周期)でのセンサ値の読み取りがほしいのかわかりませんが、
先日、Fab蔵の佐々木さんとお話した際には1msec間隔以下でセンサ値は読み込みたいといったお話がありましたので、
用途依存ですが厳しそうな感じがします。

個人的な見解ですが、
現状実装に影響を与えずに対応するには、もともとの羽多野さんspecにあったonchangeをGecko側にWorker経由で呼び出せるように実装すればとりあえずは本件回避できるので一番リーズナブルかと思います。

あくまでも個人的なイチ意見ですが。

naobsd, 2016-07-08:
@dadaa こえｒはもう実装が変わっている件についてだと思いますが、やる事としてはまだ何かあるでしょうか？