chirimen-oh/any-issues #94: Maker Faire Tokyo 2016 KDDIブース看板デザイン

kazygumi, 2016-07-06:
ギリギリとなってしまい申し訳ないです。

看板デザインを作成しましたので
先日のミーティングの意向と合っているか確認をお願いします。

[image]

[MFT2016看板.ai.zip](https://github.com/chirimen-oh/any-issues/files/350423/MFT2016.ai.zip)

gurezo, 2016-07-06:
@gurezo 個人はこのデザイン認識でいます。
強いて言えば、SPONSERSの箇所のどこか（最下部？）に
50音順、A-Z順と記載があれば、良いかと思います。

kazygumi, 2016-07-07:
@gurezo さん
ありがとうございます。
SPONSERSの最下部に「※掲載は五十音順・A-Z順、敬称略」を
追記しました。

[image]

[MFT2016看板.ai.zip](https://github.com/chirimen-oh/any-issues/files/351144/MFT2016.ai.zip)

kohichi000000, 2016-07-07:
Cの看板ですよね。
私のイメージは以下でしたがいかがでしょう？：

左（多分下側はほとんど見えない）：　上側のCHIRIMENのロゴ＋タイポだけにする
真ん中（一番見えるところ）：　上側は上記ご提案のまま(CHIRIMENロゴ＋タイポと簡単な説明）、下側に後援者、協力者を掲載する（つまり左にあるやつを移動）。
右（トークスペース側を向いている）：　何も載せないか、「Web x 宇宙 x Make」と書く

＃HAKUTOさんに確認なのですが、HAKUTOとCHIRIMENの露出は均等にしたいので。

kazygumi, 2016-07-07:
了解しました。
このような感じでどうでしょうか。

[image]

[MFT2016看板2.ai.zip](https://github.com/chirimen-oh/any-issues/files/351352/MFT2016.2.ai.zip)

ちなみにSPONSERSのところは、
横向きのA3用紙を2枚縦並びで置けるスペースを空けています。

kohichi000000, 2016-07-07:
ありがとうございます。私はこんなイメージでしたが、みなさんどうでしょうか？？

sizuhiko, 2016-07-07:
私の先日の理解では最初のイメージでしたが、看板2も良いと思います

MSakamaki, 2016-07-08:
issues整理したときにタグ我内のでeventタグ追加
あとマイルストーンとかも無いので置きました。

kohichi000000, 2016-07-08:
皆さん特に異論なさそうなので、とりあえず案２で提案します。
スポンサーは皆さんからいただいたものを入れますね～（やっぱりスポンサーがたくさんそろっているとかっこいい）

kohichi000000, 2016-07-13:
イベント事務局から２点コメントがありましたので、すいませんが、以下のように対応させてください。

①イベントのルールとして、通路側への（イベントスポンサー以外の）会社名、個人名の露出はできない。
⇒（ルールということで仕方ないので）CHIRIMENブース内側（もともとの @kazygumi さん看板１案どおり）にスポンサー名を移す。
②SPONSORだと何のスポンサーだかわからない（イベントのスポンサーと混同する）ので明確化してほしい
⇒「CHIRIMEN CONTRIBUTOR」と変更（実際に手を動かしてくださっている方という意味もこめて）

kazygumi, 2016-07-13:
わかりました。夜に対応します。

kazygumi, 2016-07-13:
取り急ぎ対応しましたー！
こちらで問題がないか確認をお願い致します〜

[image]

[MFT2016看板3.ai.zip](https://github.com/chirimen-oh/any-issues/files/361089/MFT2016.3.ai.zip)

kohichi000000, 2016-07-13:
@kazygumi さん、仕事が早いですね（笑）
これ送っておきます。

kohichi000000, 2016-07-20:
Slackにもかきましたが、ブースのロゴ掲載案です：https://www.dropbox.com/s/5qqjdextriaro0d/Maker%20Faire%20Tokyo_KDDI-0630-2.pdf?dl=0

closed 2016-08-04 by kohichi000000