chirimen-oh/any-issues #97: setup travis for MozOpenHard repo

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on December 4, 2015 6:50_

We have to run test suite on Gaia / Gecko for CHIRIMEN repo( MozOpenHard repo).

https://developer.mozilla.org/ja/Firefox_OS/Automated_testing/Gaia_integration_tests

_Copied from original issue: MozOpenHard/CHIRIMEN#35_

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on December 4, 2015 6:54_

The setup for travis was just setup by myself.

https://travis-ci.org/MozOpenHard

I'm going to check the travis to confirm to work test runner at commit or PR.
This issue is going to be still open until confirmed to work automatically test.

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on January 8, 2016 2:53_

Close
it has already DONE.

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on February 9, 2016 1:17_

reopen.
We've been get error on travis.
It seems that that error occurs on initial steps on CI.

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on February 9, 2016 1:19_

It would be better to stop the travis CI for the time being until fix that error....

naobsd, 2016-08-09:
https://travis-ci.org/MozOpenHard should/will be deleted after https://github.com/mozopenhard deletion.
it will be handled by #9 so close this issue.

closed 2016-08-09 by naobsd