chirimen-oh/any-issues #98: Port to v2.5 from v2.1

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on December 3, 2015 14:49_

## Step 1 ( only v2.5 )
- [x] Gaia 
- [x] Gecko
- [x] manifests
- [x] b2g-patches

- [x] fix bug at flash.sh ( #31, #13)

## Step2 ( v2.5 + GPIO / I2C patch / recompile kernel )
- [ ] Gaia 
- [ ] Gecko
- [ ] #45 recompile kernel

_Copied from original issue: MozOpenHard/CHIRIMEN#34_

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on December 3, 2015 15:43_

Done gaia, gecko
https://github.com/MozOpenHard/gaia/tree/v2.5
https://github.com/MozOpenHard/gecko-dev/tree/b2g44_v2_5

MSakamaki, 2016-07-08:
_From @autonome on December 3, 2015 19:33_

@YusukeYamamoto after you have ported the GPIO/I2C patch, can you attach the individual patch to https://bugzilla.mozilla.org/show_bug.cgi?id=1166588? It will give more people the opportunity to see the code and discuss how to get it landed on mozilla-central.

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on December 3, 2015 23:32_

@autonome , OK got it. 
Actually, I'm already ready to show the GPIO/I2C patch on FxOS v2.5.
ASAP, the patch will be attached on that bug.
I think, it's useful to discuss on Mozilla work week at Orlando.

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on December 4, 2015 0:29_

@autonome I've attached the patch on that bug.
https://bugzilla.mozilla.org/show_bug.cgi?id=1166588

MSakamaki, 2016-07-08:
_From @YusukeYamamoto on December 7, 2015 7:16_

I've added v2.5 branches Gaia/Gecko on MozOpenHard repo.
You can download codes with 
> git clone https://github.com/MozOpenHard/B2G.git
> cd B2G
> BRANCH=mozopenhard-v2.5-no_seccomp ./config.sh chirimen

naobsd, 2017-02-04:
20160808 release is v2.5

closed 2017-02-04 by naobsd