chirimen-oh/any-issues #100: コミュニティ名刺希望者

gurezo, 2016-07-26:
### 希望者は、下記情報を提示して下さい
- 漢字指名
- ローマ字
- 印刷部数

### 以下のいずれかの方法で連絡して下さい
- slack の 場合：@gunmaer 又は @kazy へ DM
- メールの場合：chirimen.open.hardware@gmail.com 

### コミュニティ名刺 印刷 例
- 地理麺　太郎  <== 各個人名
- taro chirimen  <== 各個人名

### メールアドレス・SNSは、下記内容が印字されます。
- mail:chirimen.open.hardware@gmail.com
- Twitter:https://twitter.com/chirimen_oh
- facebook:https://www.facebook.com/groups/chirimen/
- homepage:https://chirimen.org/

**_締め切りは基本28日の18:00とさせて頂いて、
それ以降は料金表にしたがって値上がりします。_**

**料金は下記の通り
http://raksul.com/businesscard/standard/**

satakagi, 2016-07-26:
セルフ印刷したい人向けの原稿(フリーのオーサリングツールで編集可能なもの)も提供してほしいです

kazygumi, 2016-07-28:
サンプルイメージ
[image]

gurezo, 2016-08-07:
イベントが終了したので、close

closed 2016-08-07 by gurezo

sizuhiko, 2016-08-08:
名刺代まだの人は、次回イベント（マンスリーミーティングなど）で、 @gurezo さんに渡す