chirimen-oh/any-issues #103: GPIO.write に true, false を入れた際の動作

r-konishi, 2016-07-30:
以下のURLのLチカに成功しました。
https://fabble.cc/chirimenedu/chirimenhelloworld

しかし、
```
port.write(v);
```
の箇所を
```
port.write(!v);
```
に変更したところ、動きませんでした。
結果的に "OperationError" を吐いていたのですが、なかなか気づけませんでした。

よく仕様を確認すると "numeric 0 or 1" と書いているため当たり前なのですが、
こういった場合は、コンソールを見るしかないのでしょうか。

r-konishi, 2016-07-30:
補足です。
https://github.com/club-wot/WebGPIO/blob/draft-20160125/dist/webgpio.js#L6-L9
この値が見れればうれしかったです。

naobsd, 2016-07-31:
ROWはLOWですね...

MSakamaki, 2016-08-01:
>ROWはLOWですね...

チェックありがとうございます！
typo部分だけ修正しておきました。
https://github.com/club-wot/WebGPIO/pull/15

tadfmac, 2017-02-21:
(このissueはどうなればclose?...)

dynamis, 2017-02-22:
仕様通りなので実装は問題ないし、コンソールを見るしか無いというのはデバッグはまずコンソールを見るものなのでそれを見て分かるなら十分だと思うのですが、それでも気付きにくいというご指摘かと思います。

妥当なところとしては write の引数に 1/0 以外の true/false とかを渡すと Operational Error がかえってくるという事はチュートリアルレベルのドキュメントに書いておく、もしくは仕様自体を変えて true/false (など？) も受け入れるなどにするといったところかと思います。個人的には前者推奨。

tadfmac, 2017-02-22:
GPIOの値を `true` と `false` で表現するのは違和感があるので、論理値表現である `0` と `1` で良いと思います。
チュートリアルレベルのドキュメントに書いておく、に賛成です。