chirimen-oh/any-issues #104: ちりめんの初期設定

minakokudou, 2016-08-01:
ちりめんの初期設定についてのご質問です。

ちりめんの初期設定で、以下が難しくキットに解説書などがあればと
思いました。
・基本的な導入ガイド
・配線の方法
・１分間動作が無ければ、シャットダウンする設定を変更する方法。

またスタートアップガイドなど御座いましたら是非、ご教授頂きたいです。
どうぞ宜しく御願いします。

gurezo, 2016-08-01:
@minakokudou 

導入に関する情報は、下記の通りです。
[CHIRIMEN](https://developer.mozilla.org/ja/docs/Mozilla/B2G_OS/Board_guide/CHIRIMEN)
[クイックスタートガイド](https://is.gd/7pKvD0)
[Fabble - ChirimenEdu](http://fabble.cc/chirimenedu)

[CHIRIMENの開発、ドキュメント作成に役立つツール集 ](https://github.com/chirimen-oh/CHIRIMEN-tools)

上記以外のドキュメント類は、まだ自分が知っている範囲ではないので
次回ミーティングの議題とさせてください。

kohichi000000, 2016-08-02:
クイックスタートガイド、FAQはCHIRIMEN HP配下の[こちら](https://chirimen.org/docs/ja/)
を見ていただいたほうがいいかも。
MDNは後日消されると思います。

kohichi000000, 2016-10-24:
現在のホームページがCHIRIMENを買った人がどうすればよいかわかる形になっていないように見えるので、[クイックスタートガイド](https://chirimen.org/docs/ja/quickStart.html)の構成を
・スペック編
・購入したら編（この中に、起動編、開発環境準備編、OSイメージアップデート編を入れる）
・応用編（サンプル編）
の３つにわけて、購入したら編をもうちょっと分厚くしたほうがいいと思っています。特にトラブルシューティング。