chirimen-oh/any-issues #107: CHIRIMENの名前統一

WhiteHawk-taka, 2016-08-04:
CHIRIMEN や Chirimen などのような、名前の統一がなされてない部分がTwitterなどで見られました。
CHIRIMEN に関わるページでの名前の確認が必要です。
また正式名が曖昧なのでメンバーに共有する必要があります。

gurezo, 2016-08-04:
### 現在の状況
- github
 - CHIRIMEN
 - https://github.com/chirimen-oh （URL）
- 公式サイト
 - CHIRIMEN Open Hardware（表記）
 - https://chirimen.org/ （URL）
-slack
 - CHIRIMEN Open Hardware（コミュニティ名）
 - https://chirimen-oh.slack.com（URL）
- facebook
 - CHIRIMEN Open Hardware（コミュニティ名）
 - https://www.facebook.com/groups/chirimen/（URL）
- twitter
 - CHIRIMEN Official（プロフィール）
 - https://twitter.com/chirimen_oh （URL）
- google group
 - CHIRIMEN Open Hardware（コミュニティ名）
 - https://groups.google.com/forum/?hl=ja#!forum/chirimen-oh （URL）
- gmail
 - chirimen.open.hardware@gmail.com

----

### 過去議論まとめ
githubか何かは、先に取得されて止むを得ず使用していると記憶してます。
chirimen-oh（ハイフン）/chirimen_oh（アンダースコア）問題は、
サービスの記号仕様でツイッターのみ、アンダースコア使用しています。
ツイッター以外は、ハイフンで統一としました。#9,#20,#49

naobsd, 2016-08-04:
google groupsは
https://groups.google.com/forum/?hl=ja#!aboutgroup/chirimen-oh
こっちの説明がChirimenになっています
(リンク先も直さないとですかね)

gurezo, 2016-08-04:
他にも、後でみつかりそうなので、issueも立てた事なので
次回ミーティングの再議論しましょう！

naobsd, 2016-08-08:
githubは"CHIRIMEN Open Hardware"に直しておきました

closed 2016-08-10 by MSakamaki