chirimen-oh/any-issues #109: CHRIMEN Board 発売記念イベント 企画

gurezo, 2016-08-07:
- 対象
 - CHRIMEN Board 購入者
 - CHRIMEN Board に興味のある方
- 日時
- 場所候補
 - mixi & Switch Science エイドステーション サービス利用
 - 秋葉原 イベントスペース（各種部品購入は各自）
   - 良い場所の提案大募集
- イベント内容案
 - ファームウェアアップデート・ハンズオン
 - ハンズオン
 - ハッカソン

sizuhiko, 2016-08-08:
秋葉原でのイベントスペース、永和システムマネジメントのコワーキングスペースが利用可能です。
https://www.doorkeeper.jp/%E4%BC%9A%E5%A0%B4/eiwa-systems-management

naobsd, 2016-08-08:
CHIRIMENは購入できますか?

gurezo, 2016-08-08:
スイッチサイエンスさんに残りの在庫や販売に関して、確認しないと分からない状況です。

gurezo, 2016-08-24:
> CHIRIMENは購入できますか?
イベントページ通り、10 枚？台？販売予定です。

naobsd, 2016-08-31:
なんとなく、イベントの告知(connpassへのリンク)がないなぁ、と思いました。

facebook
google calendar
slackの#sale_aniversaryのtopic
twitter

にはあるのですが、知らない・参加していない人でも見えるのはぎりぎりtwitterくらいで、それも流れてしまうような気がします。 (見落としていなければ、このissueにも書いてない、ような)

参加するような人はだいたいどれかは見ているので大丈夫、なんですかね。

MSakamaki, 2016-08-31:
connpassへのリンクは確かに無いですね、自分も良く探すのでとりあえずコメントに投稿しておきます。

http://chirimen-oh.connpass.com/event/37934/

MSakamaki, 2016-08-31:
facebookはトップへ固定できる投稿があるので、そこにサマリを常に置いてメンテするように運用するのはどうでしょう？

gurezo, 2016-08-31:
[CHRIMEN Board 発売記念イベント 情報 追記 #55](https://github.com/chirimen-oh/chirimen-oh.github.io/issues/55) サイト上に告知情報追加しました。
ツイッターの方も、日に１回カウントダウンツイートするようにします。

dynamis, 2016-09-10:
何度かお話ししましたが、先日お手伝いいただいた中央大学集中講義 (#118 ありがとうございました！) の最終日 (成果発表含む) がちょうどこのイベント日に重なっております。
両者都合が良ければということで少しお話ししていましたが、人数的に許容できるようであれば中央大学の学生 (8名 + 講師 3名) も当日同じスペースの一部をお借りして、発売記念イベントのハンズオン最後あたりに少し学生さんの発表タイムを頂けないでしょうか。(参加人数など見ながら一緒にやるかは来週金曜時点で最終確定だが基本的にはなるべくご一緒したい方向です)。

ご検討ください。

gurezo, 2016-09-10:
@dynamis さん、個人的な見解ですが現時点（2016/09/10 21:39）の
参加人数であれば、ご一緒して良いと思っております。

gurezo, 2016-09-11:
@dynamis さん、09/17は、ご一緒する場合、どのタイミングで合流でしょうか？
1. 10:00- ：ハンズオンイベント開始時間
2. 13:30-：中央大学の演習開始時間

dynamis, 2016-09-12:
タイミングについては CHIRIMEN 側のイベント二時間を揃えて 13:00 開始で考えています。

gurezo, 2016-09-12:
@dynamis さん
成果発表後に行う懇親会は、中央大学生に関して、どのようにお考えですか？
1. 会費徴収の上、参加
2. 会費徴収無料で参加
3. 懇親会不参加
4. 上記以外のパターン

dynamis, 2016-09-15:
学生さん達は 13:00 同時に開始、17:00 から成果発表をさせてもらい、懇親会については学生さんの参加自由とし、参加した人の参加費は授業側の予算で出します。
赤塚さんの方でもその方向で調整してくれていると聞きましたので、土曜はそんな感じでご一緒させてくださいませ。

gurezo, 2016-09-15:
承知しました。
現時点のconnpassの参加人数も受け入れ大丈夫だと思います。

gurezo, 2016-09-16:
部材買い出しツアーリストdone. 09/16

gurezo, 2016-09-17:
皆様、お疲れ様でした。
皆様のさぽーとに感謝です。
無事にイベントを終了したので、close

closed 2016-09-17 by gurezo