chirimen-oh/any-issues #110: CHIRIMEN-toolsの READMEリンク修正

gurezo, 2016-08-08:
TBD ... See this slideのリンク先が、MozOpenHard-monthly20151010になっています。

naobsd, 2016-08-08:
名前が問題という事でしょうか?

gurezo, 2016-08-08:
MozOpenHardが残っている事が問題と捉えています。

naobsd, 2016-08-08:
過去の議事録ですが、改竄すべきですか?
CHIRIMEN-toolsのREADME.mdにある必要があるかという事だと、たぶんnoで、必要な情報があればchirimen.orgにでもまとめるべきだとは思います。

naobsd, 2016-08-08:
どちらかというと過去(MozOpenHard)のmonthly meetingの資料を回収してmeetingリポジトリに放り込んどく方がいいような気がしてきました

naobsd, 2016-08-08:
今のREADME.mdの内容なら、単にchirimen.orgのhardware specへのリンクでよさそうです。(CHIRIMEN-toolsについての説明にすべきのような気もしますが)

gurezo, 2016-08-08:
いくつかの問題が他の要素と色々絡んで、単純なissueを立て辛い状況になっているので
時期は別途決めるにせよ、一度次回ミーティング用のアジェンダを作ります。

アジェンダの作成過程で、どの問題がどれと関連しているかを整理したいと思います。
問題やissueを簡素化したいですね。

gurezo, 2016-08-20:
[クイックスタートリンク精査、追加 #42](https://github.com/chirimen-oh/chirimen-oh.github.io/issues/42) で、 @kudodo さんと相談の上、過去の経緯を残す事になったので、 close.

closed 2016-08-20 by gurezo

naobsd, 2016-08-21:
CHIRIMEN-toolsのREADME.mdに残す事になった、という決定がなされたのでしょうか？

gurezo, 2016-08-21:
- Q. CHIRIMEN Open Hardware プロジェクト(コミュニティ） とは何ですか  
     ＝＞README.md の TBD ... See this slideのリンク先が、  
　　　[MozOpenHard-monthly20151010](https://goo.gl/2W21kB) になっている  
     ＝＞「過去にこのような議論があったよ」という意味で、上記ドキュメントのリンクを  
　　　残すという結論に至りました。

naobsd, 2016-08-21:
本issueは「TBD ... See this slide」のリンクを残すかどうかの認識です。

MozOpenHard-monthly20151010 のドキュメントを残す事、FAQなどのリンク先とする事については異論はありませんが、そのQはCHIRIMEN-toolsのREADME.mdにはない認識です。

naobsd, 2016-08-21:
すみません、その議論に参加しておらず経緯を把握していませんし、自分で手も動かせないと思いますので、やはりこのまま(クローズ)とさせてください