chirimen-oh/any-issues #113: ボードに正しいB2G OSが書き込まれていない

naobsd, 2016-08-09:
MFT2016で販売されたボードには正しいB2G OSイメージが書き込まれていません。
(何も書き込まれていない・Androidが書き込まれている・バージョン不明のB2G OSイメージが書き込まれている等、どのような状態かわかりません)

MFT2016 CHIRIMENブースにて20160429版のイメージを書き込む対応を行いましたが、20160429版には少なくとも以下の不具合がある為、現在20160808版を準備中です。

* ブートローダが不安定で書き込みに失敗する場合がある (#57)
* misc.img が間違っている (#112)
* (リリース当時の)最新のソースが反映されておらず、少なくとも Hardware Manufacturer が間違っている

テストが完了次第リリースし、イメージの更新を促す予定です。

----

ボードに正しいB2G OSが書き込まれていない件については、以下でアナウンス済みです。

- [ ] https://chirimen.org/
- [x] facebook
- [ ] google groups
- [ ] slack
- [ ] twitter

----

イメージの更新手順は以下にあります。

- [x] [Windows](https://chirimen.org/docs/ja/firmware_update_guide_for_windows.html)
- [ ] Linux

naobsd, 2016-08-09:
20160808版のテストに参加していただける方はslackの#portingをご覧ください

naobsd, 2016-08-13:
[CMN2015-1_B2GOS-20160429.zip](https://github.com/chirimen-oh/release/releases/tag/CMN2015-1) は消してしまってよいですかねぇ...

gurezo, 2016-08-13:
禍根（ブランディング的意味で）を残すので、消す方に +1

naobsd, 2016-08-15:
8/20までに反対がなければ8/21頃にCMN2015-1_B2GOS-20160429を削除しようと思います。

https://chirimen.org/ で本件について明示的な告知がありませんが、私の方では対応する余裕がありませんので、必要があればどなたかお願いします。

gurezo, 2016-08-15:
chirimen-oh.github.io リポジトリに issue を立てたので、近日中に対応します
[ファーム first release の告知 #43](https://github.com/chirimen-oh/chirimen-oh.github.io/issues/43)

gurezo, 2016-08-15:
https://chirimen.org/ 上に CMN2015-1_B2GOS-20160808 のリリース情報を公開しました。

gurezo, 2016-08-15:
https://chirimen.org/ 上に ボードに正しいB2G OSが書き込まれていない情報を公開しました。

naobsd, 2016-09-17:
CMN2015-1_B2GOS-20160429イメージとCMN2015-1タグを削除しました

gurezo, 2016-10-11:
macでのファームイメージ書き込み手順は、 #149 に引き継いで、こちらは close

closed 2016-10-11 by gurezo