chirimen-oh/any-issues #114: 起動時のロゴ(Tux)や初期化処理中に表示されるAndroidのロボットの絵をCHIRIMENロゴに置き換える

naobsd, 2016-08-09:
(TBD)

dynamis, 2017-01-20:
If you intent to change the boot logo of B2G, this bug/patch should help to understand how to do so:
Bug 1034832 - Community branding for community builds of Boot2Gecko
https://github.com/mozilla-b2g/gaia/pull/31932/files
https://bugzilla.mozilla.org/show_bug.cgi?id=1034832

gurezo, 2017-01-21:
PRを出しました
[Replacement of resource files #1](https://github.com/chirimen-oh/gaia/pull/1)

しかし、下図のように Travis CI で、エラーになっています。
 .travis.ymlが無いというメッセージですが、 Fork元にも無かったので・・・・。
リソースファイルの差し替えだけですので、判断出来る方、レビュー＆Mergeお願いします。
[image]

gurezo, 2017-01-21:
追記
リソースファイルだけなので、起動アニメーション等は
まだ手付かずです。（方法等を理解していない／分からない）

naobsd, 2017-02-04:
ペンギンロゴの置き換え、普通にやると左上からCPUの数(2)だけロゴが並ぶのですが、大きめのロゴ(480x480)を真ん中に表示するようパッチを作りました

naobsd, 2017-02-04:
リカバリの背景をCHIRIMENボードのイラストにしようかと思ったんですが、若干面倒なので(とりあえず今日は)諦めました...

naobsd, 2017-02-04:
背景を黒でなく水色 #51b5e2 にする方がよいでしょうか？

sizuhiko, 2017-02-06:
個人的には黒のままの方が好みです

naobsd, 2017-02-16:
kernel logo置き換えました

naobsd, 2017-02-16:
bootanimationの方は素材待ちですが以下の修正でいけるはず...

- https://github.com/chirimen-oh/device-rockchip-rksdk/commit/958ab8ebd68ed8987f229a0eb20d36e9c610a81c
- https://github.com/chirimen-oh/device-chirimen/commit/46db17c7f467619aad91429b3e38eceef61017ba

(現状zipは空っぽのdummy)

naobsd, 2017-02-21:
bootanimationはこれでいいはず

- https://github.com/chirimen-oh/device-chirimen/commit/705ceb88a2c84af9d82d174810eeb8c2bdd5b4d0
- https://github.com/chirimen-oh/device-rockchip-rksdk/commit/9890fd98b2a783c0ebbbc91bd0e34983d756ec6b

naobsd, 2017-02-21:
(とりあえずrecovery backgroundの事は忘れる方向で...)

naobsd, 2017-02-21:
bootanimationはmergeした

naobsd, 2017-02-21:
gaia: https://github.com/chirimen-oh/gaia/commit/1ef258c45d7a343dff80ac814dec513aa65cbe3f

closed 2017-02-21 by naobsd