chirimen-oh/any-issues #118: [9/3] 中央大学 オープンプロジェクト演習 (チューター募集)

dynamis, 2016-08-10:
Mozilla では今年、中央大学で「オープンプロジェクト演習」という 5 日間集中講義を実施します。そこでは CHIRIMEN を使って、学部一年生 (なのであまりコードがまだ書けない想定) を対象に Web 開発の入門＋ハード操作の入門を教えます。

日程としては 9/2 にイントロ (HTML, CSS, JS 基礎 1 時間含む) をして、9/3 に Mozilla Japan オフィスで CHIRIMEN で Web ページとハードの入出力を扱うチュートリアルを行います。講師としては村岡さんにメインのゲスト講師としてお願いしており、dynamis, chiko も (一部の時間抜けますが) いる状況です。もし CHIRIMEN コミュニティでもチューターとして参加頂ける方がいらっしゃればお声がけください (コメントで立候補してください)。

MSakamaki, 2016-08-10:
@dynamis  御手伝いします！

gurezo, 2016-08-10:
@dynamis 9/3は、お手伝いする予定にします

dynamis, 2016-08-24:
9/2 に事前学習させる資料 (清水さんが作成中) です:
https://mozilla-japan.github.io/open-project-workshop/

dynamis, 2016-08-24:
9/3 は taks.js をつかって main.js とか下記の gist みたいなコードをサンプルとして演習予定と聞きました。村岡さん、何か資料とかできたら、チューターしてくれる皆さんに分かるように共有頂けると幸せです。

https://github.com/chirimen-oh/WoTSignage/blob/master/js/task.js
https://github.com/chirimen-oh/WoTSignage/blob/master/js/main.js
https://gist.github.com/dynamis/0a471119b5526164cf618be6df710a1a#file-gistfile1-js-L15

bathtimefish, 2016-08-25:
はい。とりあえずLチカ、焦電センサでLチカ、httpリクエストする何か。のチュートリアルをやろうかと思っています。時間余ったらそれぞれで自由に改造してみる方向で。
手持ちのwifiドングルがchrimenに対応してないことが昨日わかったので注文中です。

bathtimefish, 2016-08-25:
焦電センサLチカのサンプルはこちら https://gist.github.com/bathtimefish/274e4103f28bd813aa921c19c0d44f70

bathtimefish, 2016-08-28:
[CHIRIMEN - Getting started.pdf](https://github.com/chirimen-oh/any-issues/files/441130/CHIRIMEN.-.Getting.started.pdf)

とりあえずスライドつくりました。これで行けるような気がします。

dynamis, 2016-09-02:
9/3 は午後とだけしか書いてませんでしたが 13:30〜18:00 になります

dynamis, 2016-09-02:
CHIRIMEN, USB Hub, Wifi Dongle, USB Mouse などお持ちの方はお手持ちのモノを持ち込んで頂けると幸いです。

bathtimefish, 2016-09-02:
中央大学は多摩キャンパスですね？

gurezo, 2016-09-02:
あれ？自分は乃木坂のモジラオフィスかと思ってました・・・。

gurezo, 2016-09-02:
> 日程としては 9/2 にイントロ (HTML, CSS, JS 基礎 1 時間含む) をして、9/3 の午後に Mozilla Japan > オフィスで CHIRIMEN で Web ページとハードの入出力を扱うチュートリアルを行います
09/03は、お昼までにMozilla Japan オフィスくれば良いのですか？

bathtimefish, 2016-09-02:
そうでしたか。では六本木に行きます。

dynamis, 2016-09-02:
あ、場所はオフィスです。オフィスに 13:00 頃までにお願いします。

gurezo, 2016-09-03:
初日終了したので、 close

gurezo, 2016-09-03:
初日終了したので、close

closed 2016-09-03 by gurezo