chirimen-oh/any-issues #125: Polyfill に関する情報について

gurezo, 2016-08-13:
### 情報の共有
Polyfill 情報が、下記リポジトリにあるので、情報の共有
https://github.com/club-wot/WebGPIO

### Polyfill 情報の扱い
- 暫定：https://chirimen.org/docs/ja/quickStart.htmlリンクを貼る
- 議論：Polyfill 情報の置き場所

gurezo, 2016-08-20:
https://github.com/club-wot/WebGPIO ではなく、
https://github.com/chirimen-oh/WoTSignage をリンク先に設定

naokisekiguchi, 2017-02-15:
[chirimenクイックスタート](https://chirimen.org/docs/ja/quickStart.html)のリンクを以下のように修正して、close

* 各種サンプル編
  * Fabble - ChirimenEdu
  * Example
  * WoTSignage
* WebI2C,WebGPIOについて 
  * browserobo
  * polyfill→polyfillレポジトリへのリンク

naobsd, 2017-05-25:
https://github.com/chirimen-oh/any-issues/issues/165 で https://github.com/chirimen-oh/WebGPIO に移動する事になったようですが、移動先は最新のpolyfillは管理できておらず、現状ファームウェアイメージと一緒に配布しているものが最新との事。

naobsd, 2017-05-29:
examplesのforCMN2015-1_B2GOS-20170301ブランチにもpolyfillが含まれている模様
https://github.com/chirimen-oh/examples/tree/forCMN2015-1_B2GOS-20170301