chirimen-oh/any-issues #128: NodeBotsとCHIRIMENのコラボイベント(もくもく会)

MSakamaki, 2016-08-15:
### 概要

 + ８月２５日 夜
 + 場所はDMM.make AKIBA
 + 参加人数は２０名前後を予想
 + 募集ページはまだ未定

### タスクリスト

 + [ ] 当日CHIRIMENの持ち出し方法
 + [x] HDMIディスプレイ
  +  DMM.make AKIBAさんで借りれる台数調整中
 + [ ] 募集ページ =>  のびすけさんのほうで用意

MSakamaki, 2016-08-16:
２５日にNodeBotsでCHIRIMENモクモク会を行う予定ですが、参加者が事前に購入すべきセンサーや、質問などを誘導するテンンプレートをイベントの概要に設けようと思っています。
とりあえず以下の内容で書いておこうと思うのですが、問題などあればご指摘ください。

---

動作確認済みのセンサーは以下にあります。
ひとまずCHIRIMENを体験したいかたは、以下のリンクにあるサンプルに記載されているセンサーや機材をお持ちください。
I2CまたはGPIOインタフェースを持ったセンサーで、サンプルにないセンサーを動作させることが出来た方は是非とも動作したサンプルコードのプルリクエストを送って頂けると助かります。
I2CまたはGPIOインタフェースを持ったセンサーで、動作しなかった物が有る場合、同リポジトリのissuesに質問として投げて頂けると助かります。

 + [動作確認済みのセンサー(サンプル一覧)](https://github.com/chirimen-oh/examples)

また、動作しない場合のissues投稿に以下の内容が書いてあるとスムーズにコミュニケーションを取る事が可能です。

 + センサー(型番)
 + 販売サイトなどのURL
 + 状況・問題
 + (可能ならば)書いてみたサンプルコード

その他質問や問題点などありましたら、CHIRIMENの技術的なサポートや質問であれば[こちら](https://github.com/chirimen-oh/Support-And-FAQ/issues)に投稿してください。
CHIRIMENコミュニティーに対しての質疑応答などあれば[こちら](https://github.com/chirimen-oh/any-issues/issues)に投稿してください。
WebGPIO・WebI2Cのpolyfill(jsファイル)に問題が有る場合は[こちら](https://github.com/club-wot/WebGPIO/issues)に投稿してください。
WebGPIOやWebI2Cの使用に問題や疑問等があれば[こちらのissues](https://github.com/browserobo)に投稿してください。

gurezo, 2016-08-18:
### 参考リンク
* [CHIRIMEN FAQ](https://chirimen.org/docs/ja/FAQ.html)
* [クイックスタートガイド](https://chirimen.org/docs/ja/quickStart.html)
* [ペリフェラル情報](https://chirimen.org/docs/ja/peripherals.html)
* [Fabble - ChirimenEdu](http://fabble.cc/chirimenedu)
* [Example](https://github.com/chirimen-oh/examples)

gurezo, 2016-08-25:
イベント開催されたので、close

closed 2016-08-25 by gurezo