chirimen-oh/any-issues #132: chirimen-oh organization に contributor team とか作る？

dynamis, 2016-08-24:
現在 chirimen-oh のスタッフは全員 organization で owner 権限が付与されています。この issues とかで個別の Issues を Assigneers にしたりするには何らかの形でチームに入っていて Write 権限とか持ってる必要があるかと思います。

今後片っ端から全員 Owner にするというでもまぁ何とかなるし、貢献は全てプルリクで受け付ければ問題ないという考え方もできますが、全体の Owner 権限を全員が持つのではなく staff とか contributors とか web-team みたいな該当リポジトリへの write 権限を付与する Team を作ってそこに入れていくのもありなんじゃないかと思います。

https://github.com/orgs/chirimen-oh/people
https://github.com/orgs/chirimen-oh/teams

dynamis, 2016-09-06:
* https://github.com/orgs/chirimen-oh/teams/chirimen-staff
  * 全リポジトリに Admin 権限付与済み
* https://github.com/orgs/chirimen-oh/teams/chirimen-contributor

naokisekiguchi, 2016-09-13:
ひとまず、contributorに
examples
chirimen-oh.github.io 
MakerFaireTokyo2016
のwrite権限を与えました。
HAKUTOさんにMFTのコードをアップしていただきたく、contributorに招待した感じです。
contributorにはどこまで付与するのがちょうどいいですかね？
他考えられるのは

- [ ] Shields
- [ ] cording-style-tests
- [ ] gecko-dev

その他どうでしょう？

gurezo, 2016-09-13:
今試作ケースを作っていますが、ケース用のリポジトリCases（仮名）を作るかどうか
思案中です。
作った場合は、上記に加えて頂けたらと思います。

gurezo, 2016-09-13:
@naokisekiguchi さんへ
下記の通り、ケース用のリポジトリCasesを作ったので、追記お願いします。
https://github.com/chirimen-oh/Cases

@dadaa さんへ
slackの連携設定お願い致します。

gurezo, 2016-10-11:
作成したので、close.

closed 2016-10-11 by gurezo