chirimen-oh/any-issues #135: コラボレーションオファーリクエストについて

gurezo, 2016-09-01:
@kskarthik さんより、コラボしないかとのオファーが来ました。
[B2G OS & Chirimen Collaboration #5](https://github.com/chirimen-oh/Support-And-FAQ/issues/5)

９月のマンスリーミーティングの議題にしたいと思います。

kskarthik, 2016-09-01:
sorry... english please

WhiteHawk-taka, 2016-09-01:
From @kskarthik , we got collaboration offer.
[B2G OS & Chirimen Collaboration #5](https://github.com/chirimen-oh/Support-And-FAQ/issues/5)
We want to the agenda of the September meeting on this topic.

naobsd, 2016-09-01:
@kskarthik this issue is just for internal communication (a note) for next local meeting. you can ignore this for now...

kskarthik, 2016-09-01:
okay!

naobsd, 2016-09-01:
@gurezo 注文をつけるばかりで手を動かさず大変心苦しいのですが、議題なら https://github.com/chirimen-oh/meeting に書けばよいのではと思いました...

gurezo, 2016-09-01:
@naobsd さん、そのとおりですね。
むやみに　issue　を作ってもしょうがないですもんね。
meeting の md ファイルに書く為のメモ的意味にもつかってしまってたので・・・。
もうちょい、工夫しますね

gurezo, 2016-09-06:
コラボレーションの議論は、https://discourse.mozilla-community.org/c/b2g-os-participation で
行なう事を提案したので、こちらは、close.

Argument of collaboration is by https://discourse.mozilla-community.org/c/b2g-os-participation.
Doing was proposed, so this issue closes.

closed 2016-09-06 by gurezo