chirimen-oh/any-issues #136: 専門学校東京テクニカルカレッジの授業協力

naokisekiguchi, 2016-09-06:
11~12月に予定する専門学校東京テクニカルカレッジでのCHIRIMEN授業について、協力してくれる方、スケジュールを決めていきたい。

naokisekiguchi, 2016-09-13:
授業のある日程は以下の通りです。
協力できる方はチェックをつけてコメント頂けるでしょうか。

- [ ] 11/4 13:30〜16:40
- [ ] 11/11 13:30〜16:40
- [ ] 11/18 13:30〜16:40
- [ ] 11/25 13:30〜16:40
- [ ] 12/2 13:30〜16:40
- [ ] 12/9 13:30〜16:40

gurezo, 2016-09-21:
@gurezo です。
直近のみ記入します。

- [x] 11/4 13:30〜16:40
- [x] 11/11 13:30〜16:40
- [x] 11/18 13:30〜16:40
- [x] 11/25 13:30〜16:40

MSakamaki, 2016-09-28:
@naokisekiguchi 調整しました

gurezo, 2016-10-11:
開発中の受講システム環境が整ったら、 @naokisekiguchi さんより、
協力者にトライアルのお願いの連絡がくる

tadfmac, 2016-10-24:
遅い時だけで申し訳ありませんが、
- [x] 12/9 13:30〜16:40

gurezo, 2016-11-02:
@gurezo です。
予定が変わってしまった為、再度書き込みします。

- [x] 11/4 13:30〜16:40
- [ ] ~~11/11 13:30〜16:40~~
- [x] 11/18 13:30〜16:40
- [x] 11/25 13:30〜16:40

closed 2016-12-27 by dynamis