chirimen-oh/any-issues #140: セルフ開発環境の構築

satakagi, 2016-09-14:
少しロングレンジの課題ですが、組み込みコンピュータとしてほとんど例がないものの、FirefoxのWebIDE相当の開発ツール、もしくはクラウドベースの開発環境をCHIRIMEN自体で動作させ、ホストコンピュータ不要のＷｏＴアプリ開発ができる環境を構築したいです。

後者（クラウドベース）はウェブブラウザベースの環境であることの強みを相当生かせる気もしますね。

kotakagi, 2016-09-14:
そういう意味で、webだけでコミットできるとか、追及していければなと

gurezo, 2016-09-15:
お話を聞いて、パッとイメージしたのが、monacaでした。
https://ja.monaca.io/

それとscrachのこれですね
http://deviceplus.jp/hobby/raspberrypi_entry_010/

皆さんは、どっちよりのイメージなんでしょうか？？？

naokisekiguchi, 2016-09-15:
monacaのイメージに近いですね。

実は専門学校さんがmonacaのアシアルさんと仲が良く、一度お話ししてます。
対象がハイブリッドなスマホアプリで、純粋なwebアプリへの対応をしていないということでした。

gurezo, 2016-09-15:
> 実は専門学校さんがmonacaのアシアルさんと仲が良く、一度お話ししてます。
> 対象がハイブリッドなスマホアプリで、純粋なwebアプリへの対応をしていないということでした。

そうすると、こっちが近いんですかね？
https://googlecreativelab.github.io/coder/

gurezo, 2016-10-11:
#147 と関連する。

gurezo, 2017-03-30:
@satakagi さん、このissue条件を教えてください

gurezo, 2017-07-01:
東京テクニカルカレッジさんで、活用したペアプログラミング環境が
KDDI リポジトリで、オープンソース化
https://github.com/kddi-research-wot/wotprogramming