chirimen-oh/any-issues #143: B2G OSミーティング参加

gurezo, 2016-09-30:
#135 の話に関連して、今週下記アナウンスがありました。
https://discourse.mozilla-community.org/t/b2g-os-and-gecko-annoucement-from-ari-jaaksi-david-bryant/11103

来週火曜日のB2G OSミーティング（日本時間10/05 AM01:00- ）の前に
CHIRIMEN Open Haradware コミュニティとしての意見や質疑応答内容をまとめたいと思います。
下記リンクに質問・要望・提案などを書き込みをお願いします。
https://public.etherpad-mozilla.org/p/chirimen-20161005

B2G OSミーティングの前にオンライン（スカイプ / apper.in）などで
CHIRIMEN Open Haradware コミュニティ有志と意見をまとめてから臨みたいと思います。

CHIRIMEN Open Haradware コミュニティオンラインMTG
日本時間10/04 PM19:30- 

参加できる方は、こちらに参加表明お願い致します。

gurezo, 2016-09-30:
言い出しっぺなので、両方参加します。

dadaa, 2016-10-03:
21 時、参加しますー

satakagi, 2016-10-04:
参加します

MSakamaki, 2016-10-04:
リモート(apper.in)から参加します。

kohichi000000, 2016-10-04:
遅れるかもしれませんが参加します（リモート）

okudako, 2016-10-04:
遅れるかもしれませんが、参加します。

marsf, 2016-10-04:
B2G OS コミュニティでは、今週はミーティングを行わないそうです。 Telegram で、今回のミーティングを提案された Karthik さんから返事をいただきました。
CHIRIMEN コミュニティ側で意見だけでもまとめておくのは、よろしいのではないでしょうか。

gurezo, 2016-10-04:
実施したので、close

closed 2016-10-04 by gurezo