chirimen-oh/any-issues #146: [12/3-4] Ogaki Mini Maker Faire 2016

tadfmac, 2016-10-03:
[Ogaki Mini Maker Faire 2016](http://ommf.iamas.ac.jp/) に、CHIIRMEN Open Hardware で申し込みました。

AIリストを立てておきます。

- [x] 応募
- [ ] 当落結果共有 (@tadfmac)
- [ ] 展示物1
- [ ] 展示物2
- [ ] 展示物3
- [ ] チラシ / パネル
- [ ] イベントスタッフ募集

kohichi000000, 2016-11-09:
@tadfmac: パネルを印刷したものが私の手元にありましたので、適当なタイミングで渡します。

tadfmac, 2016-11-11:
ホテルについて、下記連絡を頂きましたので、念のため展開します。

# ホテルご案内2通目 
Ogaki Mini Maker Faire 2016
ご出展のみなさま

こちらはOgaki Mini Maker Faire 2016実行委員会事務局です。

本日、Ogaki Mini Maker Faire 2016会場に隣接する
宿泊所のご案内をお送りしましたが、
１２月３日（土）のご宿泊については満室になりましたので
お知らせします。
同日のご宿泊をお考えの方は、近隣ホテルを
ご利用くださいますようお願いいたします。

なお、１２月２日（金）宿泊分については、
ひきつづきご予約可能です。
ご希望の方は次の電話までご連絡ください。

ワークショップ２４
　電話：0584-83-7001　受付時間8:00～18:00
　※「Ogaki Mini Maker Faire 2016の宿泊申込」とお伝えください。

# ホテルご案内1通目 
Ogaki Mini Maker Faire 2016
ご出展のみなさま

こちらはOgaki Mini Maker Faire 2016事務局です。
宿泊施設に関するご連絡をいたします。

大垣での宿泊については、市内各所のホテルに加えて、
Ogaki Mini Maker Faire 2016会場に隣接する宿泊施設がございます。
（１泊税込み１人料金 Lタイプ1700円、Mタイプ700円）
こちらをご希望される場合には、先着順で受け付けを行いますので、
次の電話番号までお申し込みください。

ワークショップ２４
　電話：0584-83-7001　受付時間8:00～18:00
　※「Ogaki Mini Maker Faire 2016の宿泊申込」とお伝えください。

■ご留意事項
　・ご予約は電話のみで承ります。
　・ご応募が宿泊可能定員を超えた場合には、受付を終了いたします。
  ・車でご来場される場合には、ソフトピアジャパン・センタービル地下駐車場をご利用ください（有料）。

■施設概要
　施設名：ワークショップ２４　７階　ソピアキャビン
　住所：岐阜県大垣市今宿6-52-18
　部屋数：全50室（すべてシングル）
　　　　　うち完全個室10室（１泊１人1700円・税込み）
　　　　　　  簡易個室40室（１泊１人 700円・税込み）
　施設紹介webページ：http://www.softopia.info/syukuhaku/
  施設の場所：https://goo.gl/maps/eMpsihu8H9P2
　※この施設はOgaki Mini Maker Faire 2016実行委員会メンバーが運営しています。

■前夜祭のご案内
　開催前日（12/2・金）には、前夜祭としてトークセッションを開催します。
　金曜日からご宿泊の方はぜひご参加ください。

　　～Ogaki Mini Maker Faire 2016 前夜祭トークセッション～

　　　著名なSF作家であると同時に黎明期から日本における
　　　メイカームーブメントに参加してこられた野尻抱介先生を
　　　お迎えして、IAMAS小林茂教授と共に今までを振り返りつつ
　　　多様性と可能性を探ります。

　　　日時：12月2日（金） 19:00 - 21:00
　　　会場：ソフトピアジャパン センタービル 1F セミナーホール
　　　　　　（Ogaki Mini Maker Faire 2016会場の１階です）
　　　入場料：無料

以上、よろしくお願いいたします。

tadfmac, 2016-11-24:
メール見落としており、展開が遅れ申し訳ありません。
12/3の夜、レセプションが開催されるようですが、申し込みの期限が本日のようです。
以下、メールコピペ

### メール
Ogaki Mini Maker Faire2016実行委員会事務局からのお知らせです。
12月18日にメールでご案内したとおり、
レセプションの参加申込みの締切が明日11月25日(金) 15時までとなっております。
ご参加される方はメールにてお申込みください。

 日時：12月3日（土）18時30分から20時00分まで
 会場：ソフトピアジャパンセンター3Fソピアホール（ホワイエ）
 参加料：一人2,000円（当日会場にてグループ毎にお支払いください）
 申込み方法：エントリーフォーム：https://goo.gl/forms/Q2ZLboEvtrdJfadO2に必要事項を入力いただき、メールでお申込みください。
既にお申し込み済の方は再送の必要はございません。

Ogaki Mini Maker Faire 2016実行委員会事務局

tadfmac, 2016-11-26:
11/28のミーティング向けに、issueを書き換えました。

WhiteHawk-taka, 2016-11-26:
12/3,4参加します。
3日の早朝に岐阜に着き、4日の晩に @gurezo さんと帰路につく予定です。
当日CHIRIMENキット一式持っていくので何かあれば言ってください。

tadfmac, 2016-11-28:
11/28 打ち合わせアジェンダ：
- 当日参加者の連絡先交換
- 展示品の簡単な共有
- 備品の確認及び回収
- 出展マニュアルの確認のお願い
- プレゼンの紹介
- 疑問点等あれば共有

WhiteHawk-taka, 2016-11-30:
作品：よくある温度をツイートするあれ(CHIRIMENでは初？)を持ち込みますのでスペースがありましたら乗せてください

tadfmac, 2016-12-01:
@WhiteHawk-taka さん
おお、ありがとうございます！
スペースは現場で調整しましょう！いかようにでもなると思いますので、ぜひ展示をお願いします。
作品の名前が決まっていたら、今からネームプレートを作成しますので、教えてください！
よろしくお願いします。

tadfmac, 2016-12-01:
Ogaki Mini Maker Faire 2016
出展者のみなさま

こちらはOgaki Mini Maker Faire 2016実行委員会事務局です。
本イベントの開幕も、いよいよ今週末に迫って参りました。
出展準備も佳境を迎えていることと存じますが、
どうぞお体に気を付けてお過ごしください。
会場内は暖房を入れていますが、１階での展示の方
（ブース番号１～30）は、出入り口からの外気が入りやすい
環境でございますので、暖かめの恰好でブースに詰めていただくと
よいかもしれません。

さて、事務局より２点お知らせがございます。

■前夜祭トークセッションのご案内
　12月2日（金）19:00-21:00に、OMMF2016の会場である
　ソフトピアジャパンセンタービル 1F セミナーホールにて
　前夜祭トークセッションを開催いたします。

　ゲストにSF作家の野尻抱介氏をお招きし、OMMF2016総合
　ディレクターの小林茂と共に、娯楽イベントではないMaker Faireの
　側面にスポットを当て、Maker ムーブメントのこれまでと、
　今後の展望を語ります。

　２日（金）より会場入りされる方、また前日入り可能な方は
　どうぞお誘いあわせのうえお越しください。
　http://ommf.iamas.ac.jp/talksession

■安全確認のお願い
　本イベントでの事故防止を図るため、12/3及び4の両日、
　係員が各ブースにお邪魔して、展示状況の確認を実施させて
　いただく予定です。
　具体的には、
　・申請したワット数以内で電気をご使用いただいているか
　・発熱物がある場合、可燃物に近接して設置していないか
　・発火や爆発するような危険物はないか
　といった点を確認いたします。
　イベントの安全確保のため、お手数ですがどうぞご協力
　くださいますようお願いします。

それではスタッフ一同、皆さまを会場でお待ちしています。

tadfmac, 2016-12-07:
ご協力いただいた皆様、ありがとうございました。
ommf無事（?ではなかった部分もありますが）終了しましたので、一旦Closeいたします。

closed 2016-12-07 by tadfmac