chirimen-oh/any-issues #147: CHIRIIMENボードのOSについて

gurezo, 2016-10-04:
### CHIRIIMENボードのOSをB2G OS以外にした場合

- Android OS系
- デスクトップ(Linux)OS系

上記の通り、OSを選定する

gurezo, 2016-10-11:
OS選定の第一弾

Picuntu on RK3066
http://picuntu.g8.net/

picuntu-mtd-visio-update-5.1-desktop.img
https://drive.google.com/file/d/0B7Qr11ucOGchX3U3QmpXN2UzZDA/edit?usp=sharing

gurezo, 2016-10-11:
時間がある時にOS選定の知見を広める。

gurezo, 2016-10-11:
### 評価項目

|OS|ネットワーク|GPIOI2C|ビデオ|カメラ|備考|
|----|----|----|----|----|----|----|
|Picuntu|未|未|未|未|イメージを焼いただけ|

gurezo, 2016-10-11:
必要な項目は適宜追加

gurezo, 2016-10-11:
#140 に関連する

tadfmac, 2016-10-13:
ちょこっと追加しました。
※ CHIRIMENのGPIOピンにアサインされている機能を想定＋αして追加しましたが、例えばUARTやAudioなどはGPIOピンではないIF（HDMIやUSBなど）を用いた実現でもいい気がします。
OSの評価結果を書き込む際には、上記も考慮して（補足しながら）記載いただけたら助かります。

naokisekiguchi, 2016-11-14:
少し試してみたことを。
http://wiki.radxa.com/Rock/prebuilt_images
こちらの[kitkat_r2_140909_nand.zip](http://dl.radxa.com/rock/images/android/radxa_rock_android_kitkat_r2_140909_nand.zip)をダウンロードして、Ubuntu 16.04.1 LTSからUpgrade_Toolを使用して書き込もうとしてみたら、
Check Chip Fail
とエラーが出ました。
CHIRIMEN初期開発時のAndroidイメージが手元にあったのでそちらを書き込んでみたら、書き込めて、androidの起動はできました。
USBハブ経由でWiFiドングルとカメラをつないで、WiFi接続の確認とカメラアプリの起動をしてみましたがどちらも使えませんでした。（昔動作確認していた頃にWiFi接続は試した気がするけど気のせいだったのだろうか）

dynamis, 2017-01-20:
Picuntu (Ubuntu 系) など Linux の検討もされているようなので FYI:

Renesas RZ/G, NXP i.MX6, TI AM572x などの SoC 向けの Minimal な Linux ビルドで Firefox を動作させる事に関する情報は GEM にあります:
https://github.com/mozilla-japan/gecko-embedded/issues

取り敢えず Cairo などあれば Linux で簡単に Firefox は動作すると思いますが、うまくグラフィックスドライバとそれ用のビルドオプションなどを選んでおかないと Firefox は GPU を使わないソフトウェアレンダリングになってかなりパフォーマンスが落ちます。また、ビデオデコード (on GPU) や WebRTC/WebGL などが動かなくなることもあります。

GPIO/I2C 周りの利用には関係ないですが、B2G からの Regression がその辺りに発生する可能性を念頭に比較検討しておくとよいかもしれません。

naobsd, 2017-02-18:
ユースケースが知りたいです

gurezo, 2017-09-30:
echigo.rev1で、B2G OS以外で、使えるOSがないかのリサーチです